Ultimaker/Cura #19081: Ultimaker 2 print outside object

PKSMANIAAAAA, 2024-05-15:
### Cura Version

5.7.1

### Operating System

Win 10

### Printer

Ultimaker 2+

### Reproduction steps

Slice the attached STL
Print

### Actual results

The print fails around 90% to the end of the print; the printer starts to print outside the object.
The issue is not occurring when using Prusaslicer.

### Expected results

Print successful

### Add your .zip and screenshots here ⬇️

[C41.stl.zip](https://github.com/Ultimaker/Cura/files/15324970/C41.stl.zip)

GregValiant, 2024-05-15:
Thanks for the report.
I don't see anything in the gcode that would cause a print to stop.  
I understand that a PrusaSlicer file finishes, but you may have to consider that something going on with the printer itself.  If this was something that Cura was doing, no one would use it.
Are you printing via an SD card, over the USB or by some other means?  Do you have any post-processors running?
When you say "...the printer starts to print outside the object." that would be consistent with a layer shift.  Is the printer physically stopping before the end of the print, or is it finishing the print but it's a failed print?

We would need a project file to look so that we have your settings.

GregValiant, 2024-05-16:
When there is a bad memory sector on a card, then yes, the print will be going along then the part of the file that is in the bad sector is ignored and the print can continue.
A weird one is when the printer gets to the back sector and jumps to a different file.  You get a different model on top of the model you started with.

I'll go ahead and close this as it was a card issue.

closed 2024-05-16 by GregValiant