Ultimaker/Cura #19082: Slicing failed with multiple models

Jolnear, 2024-05-15:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

Creality CR-10

### Name abnormal settings

Fails with default

### Describe model location

I have moved it around, rotated it, etc.

### Describe your model

I can slice the model as a single but when I multiply or bring more then 1 model in it fails.  Yes its watertight

### Add your .zip here ⬇️

I can slice the model as a single but when I multiply or bring more than 1 model in it fails. Tried running in admin mode, turning off antivirus, moving/resizing model, changing slicing settings.  I don't know if its relevant I have Nvidia 4070 and i9 CPU with 64 gb ram. Idk if its a me issue but would really like to be able to print more models at once then 1. 

https://we.tl/t-EcGlbnohEz
[cura.log](https://github.com/Ultimaker/Cura/files/15327306/cura.log)

Jolnear, 2024-05-15:
If you need anything else from me let me know tried to include everything just really wanna get this working. Also tried older versions just in case and deleting all cache files.

GregValiant, 2024-05-15:
Thanks for the report.
The project sliced for me (Win 10 Pro) and I don't find any errors in the models.
The log file has a lot of these errors: _"Passing incompatible arguments to C++ functions from JavaScript is not allowed."_
I'm not familiar with that error.

The Cura team will take a look.

Jolnear, 2024-05-15:
Thanks no idea what it could be I have tried re-instal and all that.

rienex, 2024-05-16:
Same issue with me after an win 11 update Yesterday. Before thebwin update I have no problems with that.

Jolnear, 2024-05-16:
Okay update for yall. I updated my windows cause Rien mentioned that and it sliced. But hopefully something useful in there for cura team future stuff maybe.

Jolnear, 2024-05-16:
I updated my windows  seemed to get it to kinda work it sometimes still failed but got some successful slices. I can try to reproduce it in a day or two and update the log maybe that will help.

Jolnear, 2024-06-11:
Re-coming to this topic as it has started failing again. No major changes to my PC or anything like that (newest windows update maybe?)

Jolnear, 2024-06-11:
[cura.log](https://github.com/user-attachments/files/15780835/cura.log)