Ultimaker/Cura #19084: Interface bug

4ebyrek99, 2024-05-16:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

Flying Bear Ghost 6

### Reproduction steps

When I open any panel with settings, I get an interface bug and all the text looks like random images.

### Actual results

[image]
[image]

### Expected results

Absence of any bugs in the interface display

### Add your .zip and screenshots here ⬇️

None

GregValiant, 2024-05-16:
Thanks for the report.
I've never seen that before so it might be system specific.  Did previous versions of Cura work?
Tell us more about the computer you are using (manufacturer, processor, graphics system, memory).

4ebyrek99, 2024-05-17:
Windows 10 64bit, i5-11600k, RTX 3070, DDR4 32GB 3000mhz
version > 5.6 does not work

GregValiant, 2024-05-17:
Thanks for that.
One more thing - what language are you using on that computer?  There was a mention of non-Latin based characters causing some side-effects with Cura, but I can't remember the specifics.

4ebyrek99, 2024-05-17:
I use only 2 languages in the Windows layout - Russian and English

GregValiant, 2024-05-17:
Cyrillic was one of the problems.  I did a quick search.  See bug report #18820.  That person was able to fix the problem.

The Cura team will take a look.