Ultimaker/Cura #19085: Optimize Wall Printing Order forgets outer walls for holes

Block137, 2024-05-16:
### Cura Version

5.7.1

### Operating System

Windows 11

### Printer

Custom

### Reproduction steps

1. Slice models with internal geometries
2. Enable Optimize Wall Printing Order
3. Slice

### Actual results

(Wall Ordering set to Inside to Outside)
G-code prints inner walls (green) but skip outer wall before moving to inner walls of other holes, then comes back to print outer wall (red) of each geometry individually.

Works properly for non-hole geometries.

Possibly related to #19056

### Expected results

Prints inner walls and outer wall of the same geometry in 1 go

### Add your .zip and screenshots here ⬇️

[OptmzWallOrdr.zip](https://github.com/Ultimaker/Cura/files/15339681/OptmzWallOrdr.zip)
[image]
[image]

GregValiant, 2024-05-16:
Thanks for the report.
I get the same results as you in that project.
Go to "Experimental" settings and disable "Group Outer Walls".  It looks to me like I'm getting the ordering you want with that setting disabled and Optimize Wall Printing Order enabled.

That was the only change I made and you can see below that each hole is completing before the print head moves on to another.
Let us know if it works for you.
[image]

Block137, 2024-05-16:
Looks like Group Outer Walls is the cause.
That said, the actual bug is that Group Outer Walls doesn't work with non-hole walls? (screenshot2)
Also why was it decided to be enabled by default?

GregValiant, 2024-05-21:
_"That said, the actual bug is that Group Outer Walls doesn't work with non-hole walls?"_
Sure it does.
Look at the image below and you will see that the preview slider isn't at the end of the layer.
[image]

When I slice the file with "Group Outer Walls" disabled it looks like exactly what you wanted and that final outer wall is in place.
This appears to be a settings issue so I'll go ahead and remove the bug label and close this.  Let me know if you feel that is premature.

closed 2024-05-21 by GregValiant