Ultimaker/Cura #19088: conan install failed

tonylambiris, 2024-05-17:
### Cura Version

5.3.0-alpha-xmas

### Operating System

Windows 10

### Printer

Ender 3 V2 Neo

### Reproduction steps

```conan install . cura/5.3.0-alpha+xmas --build=missing --update -o cura:devtools=True -g VirtualPythonEnv```

### Actual results

```Configuration (profile_host):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.cppstd=17
compiler.runtime=MD
compiler.version=17
os=Windows
os_build=Windows
curaengine:compiler.cppstd=20
curaengine*:compiler.cppstd=20
curaengine_plugin_infill_generate:compiler.cppstd=20
curaengine_plugin_gradual_flow:compiler.cppstd=20
curaengine_grpc_definitions:compiler.cppstd=20
scripta:compiler.cppstd=20
umspatial*:compiler.cppstd=20
dulcificum:compiler.cppstd=20
cura_settings:compiler.cppstd=20
[options]
cura:devtools=True
[build_requires]
[env]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True
tools.build:skip_test=False

Configuration (profile_build):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.cppstd=17
compiler.runtime=MD
compiler.version=17
os=Windows
os_build=Windows
[options]
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True
tools.build:skip_test=False

umbase/0.1.7@ultimaker/stable: Not found in local cache, looking in remotes...
umbase/0.1.7@ultimaker/stable: Trying with 'cura'...
Downloading conanmanifest.txt completed [0.06k]
Downloading conanfile.py completed [1.06k]
umbase/0.1.7@ultimaker/stable: Downloaded recipe revision 86ee69dd0b8ae26b5f95f511f50b85f6
pyarcus/latest@ultimaker/testing: Not found in local cache, looking in remotes...
pyarcus/latest@ultimaker/testing: Trying with 'cura'...
Downloading conanmanifest.txt completed [0.06k]
Downloading conanfile.py completed [0.15k]
pyarcus/latest@ultimaker/testing: Downloaded recipe revision 3252d78da79c9852128597be22f8d9b0
pysavitar/latest@ultimaker/testing: Not found in local cache, looking in remotes...
pysavitar/latest@ultimaker/testing: Trying with 'cura'...
Downloading conanmanifest.txt completed [0.06k]
Downloading conanfile.py completed [0.16k]
pysavitar/latest@ultimaker/testing: Downloaded recipe revision 255fd7420f9bd2208e85ac97138c1e72
pynest2d/latest@ultimaker/testing: Not found in local cache, looking in remotes...
pynest2d/latest@ultimaker/testing: Trying with 'cura'...
Downloading conanmanifest.txt completed [0.06k]
Downloading conanfile.py completed [0.16k]
pynest2d/latest@ultimaker/testing: Downloaded recipe revision b05c6e19005f46d52d9287939f1906ad
uranium/latest@ultimaker/cura_9879: Not found in local cache, looking in remotes...
uranium/latest@ultimaker/cura_9879: Trying with 'cura'...
ERROR: Unable to find 'uranium/latest@ultimaker/cura_9879' in remotes```

### Expected results

Command should've completed.

### Add your .zip and screenshots here ⬇️

![image](https://github.com/Ultimaker/Cura/assets/14058972/8de204a1-5491-49c5-8a0d-bbf88b5ce411)

tonylambiris, 2024-05-17:
[image]

Here's the actual screenshot in case it's not rendering.

GregValiant, 2024-05-17:
Thanks for the report.
That was the alpha build.  Have you seen the problem when installing a later build?  I haven't seen this problem here previously and the current Cura version is 5.7.1.

tonylambiris, 2024-05-18:
I haven't tried a later build with Cura... I had been using 5.3.0-alpha+xmas for the past year with absolutely no issues. I installed 5.7.1 and I wanted to compare the differences between the two versions because I had seen that the new trees were merged in to 5.4.0 if I remember correctly. I am having under extrusion issues that aren't easy to replicate so I was just trying to see if it was my settings or the changes in releases.