Ultimaker/Cura #19091: [BUG] [5.7.1] Artefact on horizontal surface

yet-another-average-joe, 2024-05-17:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

Voron 2.4

### Reproduction steps

Slice the provided STL

### Actual results

Cura 5.7.1 produces a artefact ; see images
Cura 5.7.0 and Orca don't

Appears with 0.2 mm layer height ; Cura 5.7.1 behaves as expected with 0.16 and 0.24 

### Expected results

a smooth horizontal plane

### Add your .zip and screenshots here ⬇️

[image]

[image]

[image]

[Pièce1 v6.zip](https://github.com/Ultimaker/Cura/files/15349275/Piece1.v6.zip)

GregValiant, 2024-05-17:
Thanks for the report.
I've seen that before and it's always been because of a slight angle on a surface, or the model isn't quite flat on the build plate.
I sliced that project file and cannot duplicate the problem.
I checked the model using MS3D Builder and the model looks good and that surface does look planar-parallel to the base surface.
The image below is at the project layer height of 0.20.  I also sliced it with layer heights of 0.30 down to 0.05 and couldn't duplicate the problem.
I see in the preview that Cura is putting an outer wall and the correct count of inner walls around that triangle.  Somehow Cura is seeing that triangle as being taller than the rest of that surface.  Since the surface looks good my best guess is that somehow a very small angle was introduced into the Cura setup causing that aberration.  I'm confident there is no problem with the model itself.
[image]

The Cura team will take a look.

yet-another-average-joe, 2024-05-17:
Really weird. Now unable to reproduce the issue.
Been thinking of some rounding errors piling up after rotating the model in various positions before repositioning it : nada !
Weirdest thing is it appeared only there. Given the length of the model (355 mm), we could expect such artefacts at others places if it was at an angle, or amplifying it with tinier layer heights.
Even weirder : just opened the attached project file : nothing !

Apart from this tiny artefact, the part came out just perfect.

Closing the bug report.

GregValiant, 2024-05-17:
I tried to duplicate this by starting out with the model as it would be when you imported it.  I flipped it all over and then tried angles as low as 0.08° I couldn't get only that corner to show.

If this comes up again let us know and this can be re-opened.  I'll leave the bug label on this but go ahead and close it.

closed 2024-05-17 by GregValiant

OsmosisD, 2024-07-12:
Happening to me too, in 5.7.2. Seems to be some sort of weird rounding error. May be hard to pin down. Only happens extremely close to 60% scale at 0.12mm layer height with a 3DBenchy.

[image]

Weirdly, I was printing this model with the same version, scale, and layer height a few weeks ago and this wasn't happening. Tested slicing with the stock profile and printer settings and it's still doing it, also for 0.2mm layer height too now. I'll just scale to 59.999% for my prints to fix it, but still weird. One of those things maybe?

[CCR100_3DBenchy.zip](https://github.com/user-attachments/files/16190154/CCR100_3DBenchy.zip) if anyone wants to take a look.