Ultimaker/Cura #19095: Slicing completely fails with all quality settings.

yorumay, 2024-05-18:
### Cura Version

5.7.1

### Operating System

Win 11 Pro

### Printer

Voxelab Aquila (Custom Ender 3)

### Name abnormal settings

-

### Describe model location

center of build plate

### Describe your model

a standard PET bottle cap, modeled following ISBT PCO 1881 standard threading.

### Add your .zip here ⬇️

[UCP_Cap.zip](https://github.com/Ultimaker/Cura/files/15360455/UCP_Cap.zip)

GregValiant, 2024-05-18:
Thanks for the report.
I didn't have any trouble slicing the project on my Win10 system.  If you right-click on the Cura desktop item and select "Run as Administrator" you might get this to work.
After a failed slice, use the "Help | Show Configuration Folder" command and Zip the Cura.log file and post the zip folder here.  It may contain a clue to why it won't slice on your system.
I'll mark this as a duplicate.  The Cura team will need the log file to investigate.

In regards to the model - the gaps in the threads may cause them to mal-form in the print.  It's better when the nozzle can move smoothly and the gaps don't allow that.
This layer shows what I mean.  The nozzle is constantly making 90° turns and short extrusions.
[image]

yorumay, 2024-05-18:
As an admin it worked, but here's the logs nonetheless. 
[cura_log.zip](https://github.com/Ultimaker/Cura/files/15362885/cura_log.zip)

GregValiant, 2024-05-18:
Thanks for that.
I'll mark this as a duplicate.
@yorumay there are several errors in the log file relating to the CuraBlender plugin.  Is that plugin working?

yorumay, 2024-05-18:
I honestly can't remember why I have it installed. Tried to slice without it in without launching in admin mode, but it failed to slice also. Here's the log without the plugin if that helps somehow. 
[cura_log_wo_blenderplugin.zip](https://github.com/Ultimaker/Cura/files/15363032/cura_log_wo_blenderplugin.zip)