Ultimaker/Cura #19096: Slicing Failed on Chromebook

Nbox2014, 2024-05-18:
### Cura Version

5.6.0.

### Operating System

ChromeOS

### Printer

Ender 3/V2

### Describe your model

The model does not matter, they all fail.

### Add your .zip here ⬇️

Last time I used Cure 5.6 was a month ago without issue. Now every model has the error "Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker." I tried to load older versions of Cura,
 and I got the same error. I tried to load 5.7 and 5.7.1 but they both crash when loading UI.
[CE3E3V2_Catfood_Can_Lid_3oz_v11.zip](https://github.com/Ultimaker/Cura/files/15360975/CE3E3V2_Catfood_Can_Lid_3oz_v11.zip)

smartavionics, 2024-05-18:
Hi, I can't help with Cura 5 but my Cura 4 based x86_64 release can be used on Intel Chromebooks. You can find it at https://github.com/smartavionics/Cura/releases. I just tried it on chromeos release 124 and it worked fine.

Nbox2014, 2024-05-18:
Thanks for the help I loaded 4.13.1 and it works. Kind of a bummer I had to go back so many versions as I had been using Cura 5 for some time now, 5.5 was the last version of Cura that worked for me without any issue. The failing to slice issue must have something to do with Chrome 124 update as it was working on 123. Hopefully, Chrome 125 fixes these issues.

>