Ultimaker/Cura #19100: Encountered a bug when slicing - not a very complex model

sgarre13, 2024-05-18:
### Cura Version

5.7.0

### Operating System

Windows 10

### Printer

Ender 3 V3 SE

### Name abnormal settings

None

### Describe model location

Centred, tube point upwards to minimise supports

### Describe your model

Port reducer

### Add your .zip here ⬇️

[Part Studio 1 - Reducer Reflex Port.zip](https://github.com/Ultimaker/Cura/files/15366902/Part.Studio.1.-.Reducer.Reflex.Port.zip)

sgarre13, 2024-05-18:
Weird this model works when the tube is horizontal, but this orientation is inefficient for supports. Hence having the tube vertyical, but Cura baulks at it - no explanation given

GregValiant, 2024-05-19:
Thanks for the report.
The model file doesn't really help.  It only tells us if the model has errors that might keep it from slicing.
We need a project file ("File | Save Project") and the Cural.log file *"Help | Show Configuration Folder").

closed 2024-06-02 by github-actions[bot]