Ultimaker/Cura #19101: One at time print sequence layer gap

Spacechicken85, 2024-05-19:
### Cura Version

5.7.0

### Operating System

Windows 11

### Printer

Custom FFF printer, no firmware modifications

### Reproduction steps

1 Place two or more objects* of different heights on build plate
* The parts need to be slightly large/complex as test cubes did not produce the same result
3 Set print sequence to One at a time
4 ensure the shortest object gets printed first, manually set it to be first if needed

### Actual results

A layer gap in the taller object at the height of the shorter object appears. This is not a visual glitch, I have observed it occurring during the print, resulting in a failed print.

[image]

### Expected results

No layer gaps should appear, and a normal print should occur. This result is achievable my manually selecting the taller object to print first.

### Add your .zip and screenshots here ⬇️

[One at time print sequence layer gap.zip](https://github.com/Ultimaker/Cura/files/15367492/One.at.time.print.sequence.layer.gap.zip)

GregValiant, 2024-05-19:
Thanks for the report.
This one is pretty bizarre and I can duplicate the problem with the project file.
I can also confirm that the Gcode file is missing the entire layer 95 from each of the two taller models.

@Spacechicken85 For now you can fix the problem by adjusting the height of the short piece.  In the project that short part is 28.6524 tall.  Adjusting just the Z and making the model 28.66 tall (and dropping it back onto the build plate) fixes the problem.
The Cura team will take a look.

@HellAholic here is an interesting one.
This is with the height change and layer 95 is present and accounted for in both of the taller models.
[image]

HellAholic, 2024-05-21:
Thanks for the report and the flag. Added a ticket for investigation.
Internal reference: CURA-11915

euf0ria, 2024-06-16:
Cura 5.7.2 here.
I have the exact same problem, but...
If i change profile to "Standard Quality - 0,2 mm" and hit slice the gaps go away.  Changing back again to my own profile brings back the gaps. 
Is there any way of comparing two profiles? The way I see it the answer lies in one or some of the profile settings but this is like looking for a needle in a hay stack.

GregValiant, 2024-06-16:
I've never found an easy way to check one profile against another.  That doesn't mean that one doesn't exist.  Maybe someone else knows a way.

At the end of any gcode file is a section called "Settings".  It lists the _differences_ between the base profile (for example "Draft") and the settings that were actually used in the slice.  It is hard to read because it is in a special format that allows you to "import" a profile by _importing_ a Gcode file.  You might be able to do something with that by checking the "Save or Discard" dialog when you switch between profiles.

I attempted a pull request on this post-processor but it was deemed to hard to maintain because between Cura versions there are always a few settings that come and go.  I think it might have some value for comparing two slices.
Unzip the folder and then copy "AddCuraSettings.py" to your configuration folder ("Help | Show Configuration Folder") and the "scripts" sub-folder.  When you restart Cura it will be available under "Extensions / Post Processing / Modify Gcode" and then "Add a script",  When enabled it will add around  310 Cura settings to the Gcode file right after the Ending Gcode and before the Cura "Settings" section (the exact number depends on number of extruders and how many post-processors were running).  You could open two gcode files in a text editor and compare them side-by-side as long as the post processor was running when each file was created.

There have been occasions where somehow a Custom Profile seems to have suffered some sort of corruption and caused problems with slicing.  In those cases the fix was to delete the Custom profile and rebuild it from scratch.
[AddCuraSettings.zip](https://github.com/user-attachments/files/15857483/AddCuraSettings.zip)