Ultimaker/Cura #19104: No model loading on Ubuntu 24.04

lalmat, 2024-05-19:
### Cura Version

5.7.1

### Operating System

Ubuntu 22.04 LTS

### Printer

Ender 3v3 SE

### Reproduction steps

Under Ubuntu 24.04 (Wayland or XOrg) :
- Launch Cura AppImage, 
- Open a STL file

### Actual results

Nothing append. 
No error message on screen.
No error message when started from command line.

### Expected results

Model loaded on platform

### Add your .zip and screenshots here ⬇️

[image]

lalmat, 2024-05-19:
Ok, seem to be solved...

When showing the about screen, the model appear just after... Very strange...

closed 2024-05-19 by lalmat

T00rk, 2024-05-20:
I have the same problem but showing the about screen doesn't work

lalmat, 2024-05-20:
Have you tried the xorg desktop ?

T00rk, 2024-05-20:
Yes I tried both Xorg and Wayland. Nothing works

T00rk, 2024-05-20:
My bad. It opens the file after 10 minutes

arsabe75, 2024-05-26:
Same issue in Kubuntu 24.04, using last app image of Cura, when I open a STL file, nothing is shown in the screen bed

T00rk, 2024-05-30:
> Same issue in Kubuntu 24.04, using last app image of Cura, when I open a STL file, nothing is shown in the screen bed

You have to wait for a loooong time. Maybe 5 or 10 minutes

sausix, 2024-06-10:
Don't understand why this issue has been closed?

As mentioned in the other issue thread, the problem is probably the `USB Printing plugin` which blocks the apllication while trying to connect to a printer.
Most people don't need that plugin since they use USB drives or Octoprint.

Can someone confirm it helps?

gtollini, 2024-06-11:
> Don't understand why this issue has been closed?
>
> As mentioned in the other issue thread, the problem is probably the `USB Printing plugin` which blocks the apllication while trying to connect to a [...]

It 100% works, you can test it yourself. Here's how:

> I fixed this by disabling USB printing altogether (I've never had an USB printer to begin with). To do that:
> 
>     * Go to the “Installed” tab of the Marketplace.
> 
>     * Scroll down to USB Printing and click "disable".
> 
>     * Restart Cura.

If you wanna know _why_ it works, here's [a quick explanation](https://github.com/Ultimaker/Cura/issues/16249#issuecomment-2095377905).

lazee486, 2024-11-01:
I dont use the usb plugin, fresh download this morning, setup ender3 pro and octoprint plugin....10 minute loads :(

fieldOfView, 2024-12-10:
The "USB Printing" plugin, not to be confused with "USB Printing NG" or "OctoPrint" plugins, is installed and enabled in Cura by default even if you do no need it. You _can_ configure Cura to disable the "USB Printing" plugin. This is independent of other plugins you may or may not have installed.

The instructions still stand:

* Go to the “Installed” tab of the Marketplace (the gear icon on the top right of the dialog).
* Scroll down to "USB Printing" and click "disable".
* Restart Cura.