Ultimaker/Cura #19105: problème lors sur tranchage

Galaxarr, 2024-05-19:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Neptune 4 plus

### Name abnormal settings

la taille pour imprimer en très grand

### Describe model location

5000% de la taille originale et le visage orienté vers l'avant de l'imprimante 

### Describe your model

c'est une figurine modélisée sur blender exportée en .stl   j'ai tenté de trancher le modèle sans succès avec le message d'erreur inattendue 

### Add your .zip here ⬇️

Je n'ai pas pu l'enregistrer avec une hauteur de couche de 0.2 j'ai été forcé de l'enregistrer en 0.24. Si il faut le fichier sur blender je peux très bien vous le transmettre.
[Uploading EN4P_rick.zip…]()

GregValiant, 2024-05-19:
Please try uploading the file again.  You might have clicked too fast when you submitted the report and the file didn't fully upload.

Galaxarr, 2024-05-19:
j(ai testé avec de nombreuses modifications mais rien n'a changé

GregValiant, 2024-05-19:
Blender can leave errors in the model.  I suspect that is the problem but I cannot download with that link and so this cannot be investigated.

Galaxarr, 2024-05-19:
Je peux vous l'envoyer le fichier blender si vous le souhaitez

Galaxarr, 2024-05-19:
[rick.zip](https://github.com/Ultimaker/Cura/files/15370489/rick.zip)
le fichier Blender ci-dessus

GregValiant, 2024-05-20:
We need the project file and the Cura.log file.  I don't use Blender.

Galaxarr, 2024-05-20:
Quel est le fichier cura.log?