Ultimaker/Cura #19107: Bridges fan speed settings

Deviaphan, 2024-05-20:
### Is your feature request related to a problem?

The fans turn on when the bridges start printing, acceleration does not occur instantly, the beginning of the bridge is printed with weak airflow. The fan turns off immediately upon completion of the bridge; if the next bridge is printed after this, the fan spin up again. This results in constant acceleration and deceleration and uneven airflow. It’s especially bad when the bridges are short - the fans just don’t have time to spin up to 100%.

### Describe the solution you'd like

Option to start the fan at 100% speed at a certain distance before the start of the bridge printing and not turn off additional cooling during short movements between bridges. This will prevent the fan from constantly turning on and off and improve the efficiency of airflow on the bridges. It would be convenient to set the maximum distance that can be printed at the "wrong" increased fan speed.

### Describe alternatives you've considered

100% fan speed whole the time, extra low printing speed

### Affected users and/or printers

This will improve the print quality of bridges on any printer, especially on printers with poor model cooling.

GregValiant, 2024-05-20:
I fool around with post-processors and there is a new one in Cura 5.7 called "Advanced Cooling Fan Control".  It's one of mine.  Something like it had been a "Feature Request" here for about 7 years.  I decided to take on the challenge even though I am "Pythonically Challenged".

There are two options in the script: "By Layer" and "By Feature" and one of the features is indeed "Bridge".  Something that became obvious while I worked on it is that for smallish models and/or fast prints, "By Layer" is a better choice for the reason you mentioned.
When using "By Feature" it takes a while for the fan motor to spool up and then to slow down.  On smallish/fast printing models you end up with an average speed rather than the set speed.  This is machine specific (depends on the actual hardware) but fan speed changes can never be instantaneous.

I would suggest that you enable the script and use "By Layer".  Look at the preview in Cura, find the layers with bridging on them, and set the fan for the entire layer.  The next setting can turn the fan off (or drop to a lower speed) once the bridging has ended.  You can have multiple instances of the script running which would allow you to mix "By Feature" and "By Layer" in the same print.  The first instance would require "Remove M106 lines prior to inserting new" to be enabled.  Additional instances of the script would need to have that setting disabled or the additional script would wipe out the changes made by the first instance.

This is an area of bridging on one of my test models.  I setup the slice so that it is a "3 layer bridge".
[image]

For that model the best way to handle bridge cooling (using the script) is "By Layer" which sets the fan speed at the _start_ of a layer.  With settings of 176/100 and then 179/0. the fan would go to 100% at the start of layer 176 and then go to 0% at the start of layer 179.  The entirety of the three layers (176, 177, 178) that comprise the "bridge" would be at 100% fan speed.  It would allow the fan to stay "at speed" rather than trying to keep up with the rapid speed changes (that "By Feature" would impose) which doesn't work so well.

With the same model stretched out to fill the bed, "By Feature" may well be a better choice.  It would depend on how the user wants the fan speed for other features like "Wall-Outer", or "Fill", etc..  The bridging would take enough time to print that the fan could get up to speed and stay there for the duration of the bridge on each layer.
[image]

Deviaphan, 2024-05-20:
> "Advanced Cooling Fan Control"

I'll try, thank you!