Ultimaker/Cura #19111: KH ForKnoR1

brdi, 2024-05-20:
### Cura Version

5.7.1

### Operating System

MacOS 11.7.10

### Printer

Flsun V400

### Name abnormal settings

Used the Ultimaker Cura profile for FLsun V400

### Describe model location

No

### Describe your model

Yes in Cura 5.3

### Add your .zip here ⬇️

[KH ForKnoRi 11.00  Hill1 6.2x4.7 Test.stl.zip](https://github.com/Ultimaker/Cura/files/15381700/KH.ForKnoRi.11.00.Hill1.6.2x4.7.Test.stl.zip)
[FV_KH ForKnoRi 11.00  Hill1 6.2x4.7 Test.3mf.zip](https://github.com/Ultimaker/Cura/files/15381709/FV_KH.ForKnoRi.11.00.Hill1.6.2x4.7.Test.3mf.zip)

GregValiant, 2024-05-20:
Thanks for the report.
I can't duplicate the problem with 5.7.1 (on Win10) as the project sliced fine for me.  I didn't come across any errors in the model.

Force the slicing error in Cura and then use the "Help | Show Configuration Folder" command to locate the Cura.log file.  Zip the Cura.log file and post that zip folder here.  Maybe it will provide a clue about what is happening on your system.

(In a minor coincidence, I just had to print a similar knob for a wall-mounted fan timer.)

brdi, 2024-05-20:
Tnx for the fast respons.I reinstalled cura and used the general PETG as material in stead of the TM3 PETG profile based on the filament I use. Now it slides as I expected. I’ll examine the TM3 profile. Must be in that profile.As for the coincidence , my knob is intended for a stove. Have to print 6 and am still struggling to print it in PETG.Bram Dijkema+31(0)6 3893 ***@***.*** NZ 21, 9805 TG  BRILTIL!- mobiel verzonden, typos may occurr -!

GregValiant, 2024-05-20:
I don't know if it's because settings get added and deleted between versions, but I've experienced problems with some profiles within Project Files that I've opened from bug reports here.  Going back to the base profile often straightens that out and then the custom profile can be rebuilt.

I print mostly PETG as my prints are generally functional items.  This is just an FYI regarding my own print settings:

- Print Speed 50mm/sec
- Retraction Distance is about maxed out at 7mm on my bowden printer.
- Retract and prime speeds at 25mm/sec.
- Print temperature 235
- Bed temperature 83
- I add 35% fan when the print gets to about 25mm tall.  I add more fan (100%) for Support Interface and for Top Skins.
- I also add fan for things like large horizontal holes as they close at the top.  That keeps the feather edges from warping upward which can cause the nozzle to collide with them.  I will also add fan at the tops of models that narrow as they get taller.

There is a new post-processor "Advanced Cooling Fan Control" that can help by allowing a user to control the fan for certain features or for certain layers in a print.

brdi, 2024-05-21:
Hi there,

Thanks for the info, i’ll experiment a bit with the settings and the Advanced Cooling Fan Control.
Consider the Issue closed.

One question:
Since I have a Flsun V400 printer, 50 mm/s seems rather slow.
I’ve printed the knob yesterday with relatively good result allthough the surface smootthness has to be improved.

CU
brdi

closed 2024-05-21 by brdi

GregValiant, 2024-05-21:
I know your printer is capable of much faster speeds than that.  The real question is - what about the material?

With the outer walls at 35 and all others at 50 the print time is 23 minutes.  At a print speed of 400 the print time is 14 minutes (with your minimum layer time of 5 seconds).  This model isn't very big so the printer isn't going to spend any time at 400.  Some infill goes down at 350 but for the most part it will be hitting the Minimum Layer Time and so Cura will drop the speed in favor of cooling.  With the minimum layer time at 10 seconds and a print speed of 400, this is a 21 minute print.  Not very much different than my 50mm/sec speed.
People are going to see these so my own view would not be "how fast" but "how good".

brdi, 2024-05-21:
Hey GregYou’re absolutely right.Finding the balance between speed & quality is the quest for the holy grail.I’ve printed the object this morning with a low layer height and 50mm/s for all speed settings and the result is perfect.brdi!- mobiel verzonden, typos may occurr -!

GregValiant, 2024-05-21:
Just because you can doesn't mean you have to.

I had noticed that when I printed silky silver with the outer walls at 35 the prints look like some sort of chrome.  If I speed up then the prints look like prints.
With most PETG's - printing at 50 to 60 gives excellent results as far as layer adhesion and structure go.  If I slow down the outer walls to 40 and with no fan most colors look like they've been polished (gray is an exception, probably has to do with the colorant).  With the cooling fan on they get a cloudy look and it isn't the same effect.
So I try to look at the entire setup and balance what the print is for.  If looks will be important, I go one way and if it just needs to perform a function I go another.
I was going to mention layer height.  There are enough curves on that model that dropping to 0.10 can make a big difference in how all the layer steps look.

For me, printing anything that has to fit on a shaft often requires a "practice" print.  I know how accurate my printer is and I tend to want to hold everything to tight tolerances.  That's often counter productive because the first one won't fit, so I end up having to print a second one to get the shaft hole the correct size.