Ultimaker/Cura #19112: Printing on 1600Yx1000x1000 printer Dowell 1610-10 DM

ArjunaJames, 2024-05-20:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

Dowell 1610-10 DM

### Reproduction steps

Using .8 mm nozzle.  Layer height 0.4 all layers.   Z hope 1.0

### Actual results

Fist layer prints, .4 layer height.  Second layer only .2 higher and causing nozzle to block. 
;LAYER:0
M106 S153
G1 F600 Z1.4
G0 F6000 X127.593 Y146.489 Z1.4
;TYPE:SUPPORT
G1 F600 Z0.4
G1 F2400 E0
G1 F3600 X115.864 Y157.183 E2.11166
[Z0=0.4 as set]

LAYER:1
M106 S255
G0 F7200 X212.638 Y251.697 Z1.8
;TYPE:SUPPORT-INTERFACE
G0 F4800 X212.638 Y251.697 Z1.6
G1 F600 Z0.6
[Z1=0.6 should be 0.8
Jamms the nozzle and stops printing]

### Expected results

Normal printing adds 0.4 to each layer.  
Somewhere betweent the Z hop of 1 mm and the next layer it forgets the Z number. 

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-05-20:
Thanks for the report.
I'm not able to duplicate this and there haven't been any other reports about layer heights being wrong.

Please include a project file ("File | Save Project") so we can see your settings.  You will need to zip the file and post the zip folder here.  If you can't share the specific model, then use another model.

ArjunaJames, 2024-05-20:
Jim Arjuna
HavenMade

Broomfield, CO
+17202824692

ArjunaJames, 2024-05-21:
What I think is happening is the Z for the supports is .3 lower than the
body of the print.  So when it finishes the Z level for the supports, it
hits the edge of the main print and is only 0.1 high causing the filament
to jamm.  This is what I think.
I am having trouble removing the supports.  So, I thought that .3 would be
OK with some sag as the body line of it prints across it.
Jim Arjuna
HavenMade

Broomfield, CO
+17202824692

ArjunaJames, 2024-05-21:
When I got to layer 95 after changing every line of code there was no "G0
line with the .3 less on the Z.

G0 F600 X521.04 Y1557.375 Z38.4
G0 F7200 X42.664 Y305.326
;TIME_ELAPSED:312208.472146
;LAYER:95
;TYPE:SUPPORT
G1 F2100 E3676.89286

This is what was on every line before Layer 95:
;MESH:NONMESH
G0 F600 X500.841 Y1557.121 *Z38*
G0 F7200 X136.476 Y742.938
;TIME_ELAPSED:311263.740240
;LAYER:94
G0 X136.476 Y742.938* Z37.7*
;TYPE:SUPPORT

I changed every line to be .3 more to match the correct layer height.
;MESH:NONMESH
G0 F600 X500.841 Y1557.121 *Z38*
G0 F7200 X136.476 Y742.938
;TIME_ELAPSED:311263.740240
;LAYER:94
G0 X136.476 Y742.938 *Z38*
;TYPE:SUPPORT
G1 F2100 E3339.22013
Jim Arjuna
HavenMade

Broomfield, CO
+17202824692

GregValiant, 2024-05-21:
We ask for project files and the Cura.log file and (occasionally a Gcode file) so we can analyze them.  The files need to be zipped (Github rule) and posted directly to the site and NOT via email (Github rule).  If you want an investigation into why you are getting the results you see then post the files.  Without them there isn't anything I can do.  I would need the original gcode file and not one that you have altered.

I changed my printer to a 0.80 nozzle and a 0.40 layer height and the gcode was exactly correct.  But that was with my settings.  I need to see your settings (project file).  As far as the gcode file goes, I've been working with gcode since my freshman year in college in 1969 and I'm pretty sure I'll be able to read it myself.

Cura never splits layer heights for any reason.   With your layer height at 0.40 and the airgap over the support at 0.30 then Cura will determine that because the airgap is less than 1 layer that the airgap will be 0 layers.  That would make it difficult to remove the support because without any airgap, the support becomes part of the model.

ArjunaJames, 2024-05-21:
I found an older version that does not do this.  It keeps the supports the
same level as the part.
Then later adjusts to allow separation of the parts.

Jim Arjuna
HavenMade

Broomfield, CO
+17202824692

ArjunaJames, 2024-05-21:
I wanted to have the space between the part and the supports at.  .3 mm.
 Because My first part with .1 mm was impossible to remove the supports
without a chisel.
On every "; LAYER:" there is a change to print supports lower. but then the
print head does not RISE up to the .8 for the second layer and consecutive
layers.  It triest to print a .1 layer height with an 0.8 nozzle, going at
60 mm per second with a volcano nozzle.  It jamms the filament, every
time.  The way to fix this is to adjust the final top layers of the
supports to the .3 difference.  I don't care about some sag I just want the
supports to come off without using a hammer and chisel.
I first printed this huge part, reduced to fit, on a 400 mm printer.  But
with a LARGE nozzle and the speed and heat needed to print faster than 2
weeks, (the first print took 14 days) I put a volcano nozzle and heat block
on.  It prints fine at over 32 mm, cubed per second.
The 5.7.1 version is good except for this problem for me.  I went back
almost a year to an older version that doesn't do this.  It remains at the
.4 difference on every layer.
The file sent here is from the Dowell version.

Jim Arjuna
HavenMade

Broomfield, CO
+17202824692
 DW_BotomHeadBraced.gcode
<https://drive.google.com/file/d/1aMhzUEvFFUKzQYOwZhR3IQWvfHnOlU8B/view?usp=drive_web>

GregValiant, 2024-05-21:
I'm going to go ahead and close this as I have nothing to investigate.  I hope you understand.

closed 2024-05-21 by GregValiant

ArjunaJames, 2024-05-21:
Every time after the first layer it jams.  It rises about .1 mm instead of
the .4 mm layer setting.
And it puts a .3 Z height lower on every layer.

Jim Arjuna
HavenMade

Broomfield, CO
+17202824692