Ultimaker/Cura #19117: Slicing Failed, 2 materials part

dd00055, 2024-05-21:
### Cura Version

5.7.1

### Operating System

Windows 11

### Printer

Ultimaker S5

### Describe your model

This is a topographic map by pieces. The other pieces printed okay but no this one.

### Add your .zip here ⬇️

[UMS5_Grid 5.zip](https://github.com/Ultimaker/Cura/files/15394308/UMS5_Grid.5.zip)

GregValiant, 2024-05-22:
Thanks for the report.
I am on Win10 Pro.  I was able to slice your project file.  I exported the model and checked it in MS 3d Builder and it appears the model has some errors.  That app is pretty good at fixing errors but it doesn't tell me what it found.   [formware.co/onlineSTLrepair](https://formware.co/OnlineStlRepair) shows this about your model.
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> **423 Non-manifold edges** (?)
--> **363 Inverted faces** (?)
--> 0 Degenerate faces (?)
--> **401 Duplicate faces** (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 53345ms ------
-> Vertex count changed from 42846 to 46486 (**+3640**)
-> Triangle count changed from 87092 to 93960 (**+6868**)

The repaired model took 5 1/2 minutes to slice.

Because of the errors - I can see where it might not slice for you.  In addition, my experience is that duplicate faces and inverted faces can make a slice untrustworthy.

I'll leave the bug label on this as it's inconclusive whether it might be a model problem or a Cura/Win11 problem.  Someone from the Cura team will take a look.

This is the repaired model.  Whether the slice is different than the un-repaired model, I don't know.
[image]

dd00055, 2024-05-23:
It's an Ultimaker Cura 5.x.x issue. I downloaded multiple Cura 5 versions,
but they all had an error. I tried Cura 4.13.1 with the "unfixed"parts, and
there was no issue.
[image: image.png]

-- 

*/ *Daniel Duque Urrego
Biomedical Engineer
Ph.D. Student - Biomedical Engineering
*Graduate Research Assistant* - AWeSOME Lab
Department of Chemical and Biomedical Engineering
West Virginia University
[image: email] ***@***.***
[image: mobile] +1 (304) 216-3305
[image: website] AWeSOME Lab <https://stephencain.faculty.wvu.edu/>
[image: linkedIn] <https://www.linkedin.com/in/danielduqueu/> [image: orcid]
<https://orcid.org/0000-0001-7924-8030>