Ultimaker/Cura #19125: Cura fills in empty space on a cylinderish Shape file

Millman28, 2024-05-23:
### Cura Version

4.8

### Operating System

Windows 11

### Printer

Elegoo Neptune 4 plus

### Reproduction steps

I dragged the file over and it did great and look great in preview. When i hit slice it filled in the middle space which was just fine in the preview and now is filled solid. 
[image]
[image]

### Actual results

Issue is in "Reproduction steps"

### Expected results

Issue is in "Reproduction steps"

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2024-05-24:
Thanks for the report.
Without a project file we can't tell what's going on.  There might be an inverted surface in the model, or you might have "Remove all Holes" turned on.  There aren't many other possibilities and this isn't likely to be a bug.
If it's the model - you can download "MS 3D Builder" from the Microsoft store.  It has a repair utility that is pretty good.

I'll leave the bug label on this for now.  If you post a project file it will tell the tale.

Millman28, 2024-05-25:
Thanks for the reply.

I have provided two files that do the same thing. The file that is 2.0 is the one shown in the picture. They are both .stl and since I can't attach them here, I put a link that takes you to a drive that has the two files.

https://drive.google.com/drive/folders/1xBzICqUcheONubPk3un7GmjY_O-dq8F1?usp=sharing

Thanks :)
:)

Asterchades, 2024-05-25:
Both files are exactly as GregValiant has suggested: the internal surfaces are inverted, meaning they're defining a solid cylinder rather than a hollow one.

[image]
This is a capture from SketchUp Make 2017, which I have configured to show "internal" faces in a burgundy colour to make them stand out more. The other model shows the same fault.

GregValiant, 2024-05-25:
@Millman28 I'll remove the bug label and go ahead and close this.  This is a popular modeling problem that can occur with any CAD application (but some are more prone to it than others).  As Asterchades shows, it's something that can be checked before you export the model from the CAD app.
Cura has some ability to visualize these sorts of errors.  Below is the flanged  model in Cura after switching back and forth to the x-ray view.  The red surface is "inverted" and the polka-dot design also indicates the error.  Cura (and PrusaSlicer and OrcaSlicer) get confused as to what is "outside" and what is "inside".  As a consequence, your ring might slice as a bowl, or as a complete solid.
[image]

closed 2024-05-25 by GregValiant