Ultimaker/Cura #19126: Ultimaker Method X not able to use single extruder full build volume

rfrischmuth, 2024-05-24:
### Cura Version

5.7.0

### Operating System

Windows 10

### Printer

Ultimaker Method X 

### Reproduction steps

- loaded a stl part with 184 x 184 x 10 mm
- deactivated second extruder

### Actual results

unable to slice because part is too large,

### Expected results

should be able to slice because build volume with single extruder is 190 x 190 x 196 mm (see specs for single extrusion https://store.makerbot.com/3d-printers/methods/method-x)

=> sliced the part with Makerbot Cloudprint and it worked..

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-05-24:
Thanks for the report.
The Cura team will need a project file to investigate.  They will need to know what your settings are.
The "Build Plate Disallowed Areas" might be coming into play as it appears that at 184 x 184 your model intrudes into the disallowed areas (as you can see in your screenshot).  It appears that the largest model would be X150 by Y187.  The Makerbot app uses a different printer definition file than UM Cura and those areas may be defined differently.
[image]

rfrischmuth, 2024-05-24:
Thanks for your quick response. See the attached project file.
[UMMX_rahmen-platte.zip](https://github.com/Ultimaker/Cura/files/15431738/UMMX_rahmen-platte.zip)

Ultimakers website defines in their specs the build volume as
Build volume (dual material)
W 152, D 190, H 196 mm (W 6, D 7.5, H 7.75 inches)
Build volume (single material) **[my case]**
W 190, D 190, H 196 mm (W 7.5, D 7.5, H 7.75 inches)

Im a bit irritated that this shouldn´t work with Cura?

Thanks.

GregValiant, 2024-05-24:
With one extruder disabled it would seem that one of the disallowed areas should go away, but it doesn't.  The build plate in Cura is a lot different than when I activate an S5.

Maybe @HellAholic from the Cura team can explain what is going on.

HellAholic, 2024-05-24:
Hey @rfrischmuth , the current implementation for the Method X/XL are the first iteration of the printers in Cura. With time those definitions will be improved by either community contributors or the new features developed by the Cura team.
As such, the first priority was at the time, ensuring a good print quality with dual extrusion as most requests were for that feature.
In future releases the update to the logic for switching between the two options could be included, but it depends on the number of requests and how much a feature is in use.
So TLDR; this is a feature that needs to be worked on. 

Apart from that, build volume often does not translate to print volume since there are multiple factors that affect the size, such as shrinkage factor, margins for raft/skirt/brim/support/etc.

GregValiant, 2024-05-25:
I'm just thinking here and one of the things I'm thinking is - it is not an inexpensive machine.

Ignoring IDEX printers, the "Tool Offset" of a dual-extruder printer usually determines the disallowed areas.  Cura assumes that neither extruder can move beyond the limits of the build plate.  That generally means that the left extruder cannot make it to the right edge, and the right extruder cannot make it to the left edge.  Those areas become the "Disallowed Areas" for each nozzle.  When a nozzle is disabled, the Disallowed Areas are adjusted on-the-fly.

Please understand that I'm barely familiar with the UltiMaker line of printers, and not at all familiar with the Makerbot designs so this might be completely off base.
Looking at images of the Method X printer, it appears that the nozzles are a fixed distance apart.  The Method X definition states that there is no "nozzle offset" and "Apply Extruder Offsets to Gcode" is checked.

@HellAholic - my question is:  Is the Gcode shift being handled in the firmware to sync the print location for each nozzle?  If so, is that why the disallowed areas are handled in the definition file?   Disallowed areas that are 66.65 wide seems to go beyond what appears to be the nozzle offset in images I've found of the Method X (which looks to be closer to 25mm).

The Sovol SV04 printer seems to face a similar situation.  The way Sovol handles that is to provide multiple definition files.
In the case of the Method X machines, should there be separate definition files for:  "Extruders Both", "Extruder 1 only", and "Extruder 2 only"?  The user would install all three printers and pick their active printer based on the intended need of a particular print.  It would negate the necessity of disabling an unused extruder, and each definition file would have the necessary defined Disallowed Areas.

An alternative might be to enter the tool offset in the definition file but to un-check "Apply Extruder Offsets to Gcode".  Would that allow the firmware to make the shift for the nozzle offset?

I have an interest in this because one of the tools I use for analysis is a macro that reads Gcode files into AutoCAD.  It appears that it would need to be adjusted for the Method printers.

GregValiant, 2024-05-25:
@rfrischmuth,
I think there is a viable workaround for this.  It is similar to something I've done for a couple of custom built IDEX printers.

Removing the left and right "disallowed areas" from the printer definition file allows the user to position the model where they want in the X.  That is usually a _bad thing_ as the printer nozzles cannot physically get to certain areas.  
Also altering the Cura printer platform so it gives a graphic representation of the disallowed areas allows the user to slice a model using more of the build plate.  It would give a "printable area" the size that the advertisements say it will be.

You can see below that I have moved your model to the left and it will now slice there and will print with the left extruder.  BUT...it would also slice if I had slid it to the right and the print would fail because in reality the left extruder can't get that far to the right.  That is the downside to a hack like this.  The user must pay attention.
Within the printer definition file, each disallowed area is 66.63mm in the X.  Whether that number is viable, I don't know.  It would take some experimentation to work out how far to the left that the left extruder will go, the distance between the extruders, and how far to the right the right extruder can go.
[image]

It would be better if UM could push out a quick fix for this.  I just thought I'd mention a workaround that could be done now.

rfrischmuth, 2024-05-27:
@GregValiant Thank you very much for your solution! That is fine for me as a workaround.
@HellAholic Thanks for your response!
In my opinion UM should fix this even if customers doesn´t report this actively as missing feature. This machine is sold by UM and should be fully supported by their own slicer software.

HellAholic, 2024-05-27:
@GregValiant TBH I am not 100% sure, that seems to be the case at the first glance but I was not fully involved in the logic process. I'll let you know if/when I find things out.
@rfrischmuth The Method line has a dedicated cloud slicer, we added the printers (MethodX/XL) to Cura to see how many users migrate to Cura (mainly for offline slicing via USB) and we will continue to add features and fix issues on them. To clarify my previous comment, adding functionality and features for the printers is on our backlog but the process involves a lot of testing and validation to ensure a good quality so features with highest requests take the highest priority.

GregValiant, 2024-05-27:
@rfrischmuth,
Here are the two files I adjusted as a workaround for this issue.
[ultimaker_methodx.def_GV.zip](https://github.com/Ultimaker/Cura/files/15455207/ultimaker_methodx.def_GV.zip)

The preferred place to copy these files to would be your configuration folder:  The JSON file goes into the "Definitions" folder and the STL file into the "Meshes" folder.  On a Win 10 system that should be "C:\Users\ ...name... \AppData\Roaming\cura\5.7".  When a new MethodX definition file is released by UM you can delete these files.  You can (of course) delete them at any time and Cura will simply revert back to the original platform and definition.
I've altered the definition file to eliminate the left and right "disallowed" areas and I've changed the "platform" to "ultimaker_method_platform_GV.stl".  That's the STL file shown in the screenshot above and is the one I've included.

**Disclaimer:**
_The Disallowed areas are there as a safety_ and this definition file doesn't have that safety.  It leaves it _up to the user_ to know how large a model will fit and print, and where it goes on the build plate.  That is true when printing with both extruders, or just with one.
The attached definition would allow you to do something like shown below, and it will slice, but IT WILL NOT WORK and the printer may attempt to get to areas where it cannot.  That is usually considered to be a **_"BAD THING"_**.  Pay attention.
[image]

I'm not related to UltiMaker in any way, just a "member of the community".  When I purchased my printer (a Creality machine) 4 years ago, the UM Forum was really the only place that gave good sound advice.  Going over the bug reports here I had to ask myself - "Heh, if this software is so bad why does anyone use it?"  It turned out that very few of the reports were bugs in the software.  Nearly 90% are either bad models, misguided users, or incorrect settings.  There has also been a fair share of incorrect definition files.  This is the first I've seen regarding a UM "branded" machine.

Between the two sites I've learned a lot and one of the things I've learned is that the folks at UM are bright, dedicated individuals.  The MakerBot merger can not have been easy.  They are different machines that work differently and have different firmware so integrating the MakerBot line into UM and Cura is going to take a minute.  This definition problem might not be the first trick I've seen UM miss...but there certainly haven't been many.  I am confident UM will address the problem shortly.

If you have any issues with what I've done let me know.