Ultimaker/Cura #19127: Slicing failed after a long time trying.

benclewett, 2024-05-24:
### Cura Version

 5.6.0

### Operating System

MacOS 14.4.1

### Printer

Enders 3 v2 Neo

### Describe your model

Never sliced yet.

### Add your .zip here ⬇️

Project file here:
https://we.tl/t-PIK6eD36Xh

GregValiant, 2024-05-24:
Thanks for the report.
Your model is loaded with errors.  Cura marks surfaces with a polka-dot pattern when they are inverted or have other problems.
I used 5.7.1 for these screenshots.  I do have the "Mesh Tools" plugin installed.
[image]

This is in "x-ray" view.  Most of this model appears to be "Invalid".
[image]

At 507mb it's a really large STL.  Both MS 3D Builder and PrusaSlicer's NetFabb worked on it for a long time and neither was able to affect a complete repair.  I think this model needs to go back to CAD to be fixed.
There isn't a way to tell if the model is keeping Cura from slicing, or if it is a bug in Cura.

This model can't be sliced by Cura, PrusaSlicer, IdeaMaker, or OrcaSlicer.  PrusaSlicer crashed during the slice and Orca slicer informed me that even after the repair by NetFabb, the model still had 630,000+ non-manifold edges.

I'm going to remove the bug label and close this.  Your model is invalid.  I hope you understand.

closed 2024-05-24 by GregValiant

benclewett, 2024-05-24:
Thanks for your email.  I will install the Mesh Tool plugin, I can see this
is extremely useful.  And return to the CAD system to find the errors.

Many thanks for your time.

Ben