Ultimaker/Cura #19130: When i click Slicing says: Slicing Failed

BLAI38, 2024-05-24:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

Anycubic Kobra 2

### Describe your model

Are 6 gears 

### Add your .zip here ⬇️

[image]

GregValiant, 2024-05-24:
Thanks for the report.
Yes, they look like gears and there are six of them.
Unfortunately, without a Project File and the Cura log that's all I can tell from here.  The instructions are on the original page you filled out for this report.

closed 2024-06-01 by BLAI38

BLAI38, 2024-06-01:
I am sending you the link of a handheld fan grip, along with the gears for this project. Also, I want to let you know that I set it at 80% of the size, just in case. Thank you.

Best regards.
Link: https://cults3d.com/en/3d-model/tool/cool-squeeze-grip-fan

BLAI38, 2024-06-04:
> I am sharing with you the hand fan grip link and gears project. Just a heads up, I've reduced it to 80% of the size. Also, I have attached the .log file. Thanks.
> 
> Best regards. Link: https://cults3d.com/en/3d-model/tool/cool-squeeze-grip-fan .log file: [cura.log](https://github.com/user-attachments/files/15524745/cura.log) "I have issues only with these files: GripFan_Wheel_2A GripFan_Wheel_2B GripFan_Wheel_3A GripFan_Wheel_3B GripFan_Wheel_4A GripFan_Wheel1