Ultimaker/Cura #19133: Cura crashes and displays fatal error message

Neal3D, 2024-05-25:
### Cura Version

5.7

### Operating System

Windows 11 home

### Printer

Makertech Proforge 4.1 with klipper

### Reproduction steps

I am trying to uninstall Cura completely so that I can install it correctly I have tried uninstalling it and deleting the files that cures website says to delete before uninstalling and it still is not erasing it.
I want to reinstall it completely clean

I tried to install Makertech Proforge 4.1 I copy the files into Cura but I think I put it in the wrong folder and now it crashes as soon as you open it and it says

Crash Report
A fatal error has occurred in Cura. Please send us this Crash Report to fix the problem

[image]

### Actual results

After you hit send bug report Cura crashes

### Expected results

It should have added the new printer profile

### Add your .zip and screenshots here ⬇️

This is the GitHub for the     Makertech Proforge 4.1  files that I transferred to Cura

https://github.com/Makertech3D/Makertech-Cura_Resources

GregValiant, 2024-05-25:
Thanks for the report.
If there is a coding error in any of the definition files it will cause Cura to crash.  It could be something as simple as a misplaced quotation mark or an unmatched parentheses symbol.  It would be up to the provider of that definition to figure out where the error is.
The last couple of lines in the top textbox of that crash dialog will often point to the file that is causing the crash.

I'm going to remove the bug label from this as it is very likely a problem with the definition files from MakerTech.  Let me know if you feel that is premature.  I'll leave this open for a bit so you can respond.

You can delete the "Configuration Folder" (on Windows is is usually something like "...C:\Users\...name...\AppData\Roaming\cura\5.7") if it was not removed during the un-install.  That should allow a clean re-installation.

Neal3D, 2024-05-25:
Thank you I'll try that and see if that fixes it

GregValiant, 2024-05-25:
I adjusted that printer definition file by fixing all the indents.  There was also an extra "}" that would definitely cause a crash.  After the repair it seemed to be OK.

closed 2024-05-25 by GregValiant

Neal3D, 2024-05-26:
I got cure uninstalled and reinstalled as if it was new. Thanks to your help!! Do you mind sending me the file printer definition file, you fixed? As you can tell, I'm not very good at this stuff.

GregValiant, 2024-05-26:
That's official MakerTech software so I won't step on their toes or create an "unofficial" version.

Any definition file involves numerous other files for Quality, Nozzles, Extruders, etc..  I took a quick look at one of them and found errors.  Did I luck out and find the only one with errors?  Did I find all the errors?  Probably not.  All 25 or so definition files would need to be gone through.

I did fill out a bug report on that Github page.

Neal3D, 2024-05-26:
That's completely understandable thank you