Ultimaker/Cura #19134: The help system unable to scroll infill pattern

PGTBoos, 2024-05-25:
### Cura Version

5.4.0

### Operating System

win 11

### Printer

cr10 sv2

### Reproduction steps

the help system sometimes its a lot of text, the issue happens at several places.
for example the infill pattern, its a pulldown, and a lot of options but the help system isn't scrolling with the options together, so essentially I cannot read the text.

### Actual results

i'm unable to read the helpfile of infills and several other options as well

### Expected results

be able to read the whole help, maybe use a fade time don't hide it directly, so I can move the mouse into it, and add a scrollbar in the help dialogs so as long as my mouse then is in the scroll area I can scroll and read.

### Add your .zip and screenshots here ⬇️

no more info

GregValiant, 2024-05-25:
Thanks for the report.
This can be changed if you have certain plugins installed.  One of them is "Settings Guide" which makes the text area of the balloon a lot larger for some settings.  Maybe the tooltip balloon isn't expanding far enough.
Do you have that installed?  It makes a difference as to where to start looking.

PGTBoos, 2024-05-26:
If the text would get larger it would get worse. The behaviour can be solved if it was coded this way:
- hoover over a setting
- causes the help to popup, don't let the popup auto close upon leaving the hoover instead keep it 2 seconds
- if within the 2 sec the mouse is within the help area don't close (ea add 2 sec).
-while in the help area allow the mouse to scroll the help area

Outlook voor Android downloaden<https://aka.ms/ghei36>