Ultimaker/Cura #19139: Really Very Slow import of .stl files

redzabu, 2024-05-26:
### Cura Version

5.7.1

### Operating System

Fedora40 Wayland

### Printer

Creality Ender 3 s1 pro

### Reproduction steps

Starting fresh cura appimage - OK - loads  in less than 20 sec ...
Load a simple testcube .stl file takes more than 90 sec on an AMD7 7 PRO 6850H with Radeon Graphics ?!?!
This behaviour is consistent across 3 machines (2 AMD 7 and 1 I5)

### Actual results

import of a simple .stl file in a few seonds

### Expected results

Import takes AGES!

### Add your .zip and screenshots here ⬇️

[image]

Asterchades, 2024-05-27:
Without the affected STL to replicate the issue there isn't going to be a great deal that can be done about this. It could be a corrupted or faulty file, or it could be an edge case which has implications elsewhere but hasn't been nearly as obvious an issue to investigate.

CrashTD, 2024-05-28:
Likely dublicate of #18968?

Check if disabling USB Printing fixes the issue. This is of course no permant solution.

redzabu, 2024-05-28:
Spot on !

disabled USB printing and ... problem solved

For now i'm happy again - Thanks !

kafufafu, 2024-06-05:
Could this have had any relation with the following? #19186

ShyneTurtle, 2024-07-17:
Fedora 40 Wayland (asahi linux, aarch64) on cura 5.4, had the same problem and disabling USB printing fixed it.

philibertc, 2024-11-10:
Hello, same issue for me on Fedora 41 with cura AppImage.
Disabling USB printing fixed it.

beta-tester, 2024-12-04:
same here, importing STL or 3MF files takes forever. in a previouse version of Cura is wasn't that an issue.

```bash
$ uname -a
Linux pc1 6.8.0-49-generic #49-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov  4 02:06:24 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

```bash
$ cat /etc/os-release 
PRETTY_NAME="Ubuntu 24.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.1 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo
```

```bash
$ flatpak list
Name                                           Application ID                                               Version                           Branch              Installation
UltiMaker Cura                                 com.ultimaker.cura                                           5.9.0                             stable              system
```

mpiscaer, 2025-03-22:
For me the same, disabling the package "USB printing" in the marketplace did the trick. Thanks @philibertc  for the tip.