Ultimaker/Cura #19145: Could not save to /AV_Tapa Tel 609 v1.gcode: [Errno 30] Read-only file system: '/AV_Tapa Tel 609 v1.gcode'

Ciruplas, 2024-05-27:
### Cura Version

5.7.1

### Operating System

mac os Sonoma 

### Printer

anycubic vyper

### Reproduction steps

Created a file in Fusion 360, exported in stl format. Opened Cura , lay it flat, sliced it and impossible to save it.
Following message " Could not save to /AV_Tapa Tel 609 v1.gcode: [Errno 30] Read-only file system: '/AV_Tapa Tel 609 v1.gcode'".

### Actual results

Doesnot apply

### Expected results

Being able to save in gcode the file

### Add your .zip and screenshots here ⬇️

.

GregValiant, 2024-05-29:
Thanks for the report.
Is the folder you are trying to save to "Read Only"?  Is there an existing file with that name and it is open in another application?
Cura makes calls to the operating system when opening or saving so this isn't likely a Cura problem.  It would seem that Sonoma doesn't like the name, or the location you want to save to.

Ciruplas, 2024-05-30:
Dear Greg,
Thank you for your prompt response . I think its something related to Sonoma  since I ´ve been using Crura and Fusion 360 before I updated Mac OS with no problem . I decided to try to save the gcode file not to the desktop but directly to an  SD card and it was saved with no error message, it just won´t save it to the desktop , documents or any other place in the hard drive  . 

Thank you !! Best  regards

Ramiro Huertas

GregValiant, 2024-05-30:
I'll go ahead and close this as the problem really can't be Cura and it sounds like everything was OK before the update to the operating system.
One of my life rules is "Never upgrade an operating system" because something won't get the memo, something will be removed that wasn't supposed to be, etc.

closed 2024-05-30 by GregValiant