Ultimaker/Cura #19146: Blobbing with klipper after filament change

FuturSole145, 2024-05-27:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Creality Ender 3

### Reproduction steps

1. Added g-code command to filament change (M600)
2. Change Filament
3. Manually Purge Filament
4. Resume Print

### Actual results

After resuming the print, the gcode makes the extruder push out too much filament therefore making a giant blob on my print

### Expected results

the print should resume without making a giant blob

### Add your .zip and screenshots here ⬇️

[XYZ 20mm Calibration Cube - 1278865.zip](https://github.com/Ultimaker/Cura/files/15459249/XYZ.20mm.Calibration.Cube.-.1278865.zip)

GregValiant, 2024-05-27:
Thanks for the report.
Having the STL doesn't do us any good.  We need a project file to see your settings and how you have configured M600.
If Cura is adding the M600 line to the gcode then there is no bug here.  M600 puts everything onto the firmware.  You need to configure Klipper to provide the correct amount of prime.

This is the total extent of the gcode added by "Filament Change":
;BEGIN FilamentChange plugin
M106 S127
M600 E30.00 U300.00 X0.00 Y0.00
;END FilamentChange plugin
That's it.  A line for the fan, and a line to tell the firmware there will be a filament change.  There is no extrusion'prime inserted into the gcode.  In this example there is an initial retraction of 30mm and an unload distance of 300mm.

If you have a direct drive extruder that uses short retraction,s and Klipper is set up to respond to M600 with a 6.5mm retraction, then there will be a 6.5mm prime when it should be somewhere around 1mm.  That will leave a blob.

I'm going to remove the bug label from this but I'll leave it open so you can respond.

GregValiant, 2024-05-29:
I have filled out an information request at MarlinFW regarding the numbers used for the M600 E, U, and L parameters.  It isn't clear to me if they should be positive or negative.  It would certainly make a difference if a 30mm retraction turned into a 30mm prime.