Ultimaker/Cura #19147: Add precise z-height setting.

T9Air, 2024-05-27:
### Is your feature request related to a problem?

Sometimes I have a model that is 1.05 high, which needs to be that high for whatever reason. however, I don't want to have to deal with setting the layer height to .105 mm high, as it would take more time, and if my model is 1.005 mm high, and it needs to be that high, I can't set the layer height to .1005, as Cura does not allow that.

### Describe the solution you'd like

what would be good is if Cura could realize that the model is 1.05 mm and the layer height is .1 mm, and therefore on the last 5 layers, automatically adjust the layer height to .11 mm.

### Describe alternatives you've considered

I have noticed that the ability to do this has been included in the most recent release of Bambu Studio (1.9.0), but I do not want to switch, as I am used to Cura.

### Affected users and/or printers

I think this will affect anyone who has a model that is just above or below the set layer height (ex: model height, 1.01 mm, layer height, .1 mm) and does not want to change the layer height for the whole model.

### Additional information & file uploads

These images will explain it better than me since they came from the Bambu Studio change log.

[image]
[image]

GregValiant, 2024-05-28:
I have had to do this on occasion.  If it is a single feature then adjusting the "Initial Layer Height" can resolve the problem.  That can also be done when adjusting for an exact height of a single intermediate feature.
But if there two or more features that need an exact height, adjusting the Initial Layer Height isn't really a viable workaround.
For your example model you could print it at .20 layer height and an initial layer height of 0.30 and you will hit your 12.3 model height.

T9Air, 2024-05-28:
if my model is 1.0001 (extreme example), I can not do this manually. Also, I do not want to do it manually, it is better if the slicer can do it itself.

T9Air, 2024-05-29:
It could be, even though it must be easier if it is separate because in Bambu Studio, it is separate.

al-65, 2024-05-29:
Actually, I would love such a feature :) My application is optics adapters (adapt old lenses to MFT mount cameras) where the height accuracy is vital for focus! see thingiverse thing 4349420

T9Air, 2024-06-17:
> It could be, even though it must be easier if it is separate because in Bambu Studio, it is separate.

Also, adaptive layers does not change height based on the model height, it changes based on the model's curves, while precise z-height changes based on the model height

T9Air, 2024-06-19:
I tried adaptive layers, and it takes a really long time to get it to the right hight, if that is even possible.

T9Air, 2024-06-21:
@GregValiant do you know where I would be able to put the code for a precise z-height setting? I am working on porting it from Bambu Studio.