Ultimaker/Cura #19149: Specify a gap between outer wall and inner (next) wall

J12345T, 2024-05-28:
### Is your feature request related to a problem?

I have been using Cura for a few years mainly for 3d printed RC aircraft. This focuses on light weight foaming pla filaments printed in a single or double wall outer shell. The product is a light weight but often fragile model.

I have been experimenting with a 2 wall outer shell print where the 2 walls are artificially separated by a small amount. For the same weight of filament this gives a significant increase in strength to the model.

I tried to trick the slicer by using the Outer Wall Inset feature and giving it a small negative value. Whilst the slicer accepted a value (-0.2mm) it did not seem to have an effect on the print.

### Describe the solution you'd like

So I designed a CAD model with a 1mm shell. I set a 0.4mm wall width and the slicer printed an inner and outer wall separated by ~0.2mm. This is how I know the method produces a stronger result.

My request is to have a setting where I can specify a 'gap' between the outer wall and the inner wall such that I can CAD a solid model and let the slicer position the 2 walls as I've described.

A second feature request would be the ability to add a fuzzy wall finish to the INNER wall only. This once again would be to enhance strength for the minimum additional filament weight.

### Describe alternatives you've considered

Outer Wall Inset as described above. A small negative value although accepted by the slicer did not have an affect on the printed model.

### Affected users and/or printers

This modification will be of benefit to the RC aircraft 3d printing community and to all who use foaming filaments to gain lightweight and strong structures.

### Additional information & file uploads

[image]

GregValiant, 2024-05-28:
I think you can do this now.
[image]

The slice is with layer height 0.20 and with the line widths at 0.40  EXCEPT the inner wall line width is at 1.0.
Then in the Material Settings I've set the Inner Wall Flow to 40%.
The extra line width moves the centerline of the nozzle over and then the reduced flow gets it back to a 0.40 line width.
This example leaves a gap between the inner and outer walls of 0.60.
I did notice that the tops and bottoms got funky but a support blocker configured as a cutting mesh with Inner Wall Line Width at 0.40 and the flow @ 100% brought everything back the way it should be.

Here is the top the way it would print with the odd inner wall line width and flow.
[image]

On the bottom I have the cutting mesh in place to get the line width and flow back to normal.
[image]

J12345T, 2024-05-28:
Hi Greg. Thank you for this. I would never in a million years found this myself. That said, there is still value I think in a specific setting to control this and possibly a fuzzy inner wall. The reason is that with foaming PLA the flow is normally set to 50-60%. In order to make your method work I need to set the inner wall flow to something like 25% which the slicer colours in amber meaning that it doesn't like the setting? I'll give it a try and see what the results yield. Many thanks again. JT

GregValiant, 2024-05-29:
That isn't how it works.  The "Line Width is just the index distance between two extrusions.  Cura then calculates how much plastic will be required to fill that area.
For an example let's say we have a 0.4 nozzle and the print is at 0.20 layer height and 0.40 line width.  The index distance between extrusions is then 0.40 and the area that needs to be filled with plastic is .08mm².
The inner wall line width of 1.0 means Cura will calculate 2 1/2 times as much plastic to fill the wider gap.  Setting the inner wall flow to 40% means 40% of 250% which comes out to an actual flow rate of (amazingly enough) 100%. 

I think it's worth doing a couple of tests.  At first glance it sounds like your flow will be low, but it won't be.  It will be the same as any other .2 x .4 extrusion, it will just be farther away from the outer wall extrusion.  You can see it in my screenshot - the extrusions of the inner wall and the outer wall are the same width.  It's just tricks with numbers to fool Cura.

J12345T, 2024-05-29:
Right I understand now. Thanks. I've run some experiments and it certainly works so I may look into it further.

Do you know if there are any possibilities of getting the fuzzy skin on the inner wall?

Cheers. JT

GregValiant, 2024-05-29:
Those calculations are within the Cura Engine and I'm not familiar with it at all.  Off the top of my head I'm going to say no, you can have "outside only" or "inside and outside".
I don't think it's anything that will be added as it is such a small niche-use.  Probably 99% of the projects that I've seen that use Fuzzy Skin are game controllers.  The other 1% are similar models that the user wants a "grip" surface on.