Ultimaker/Cura #19155: Unable to slice an Object with a Z height of .20mm ie a 1st layer print

MikeOxawopper, 2024-05-28:
### Cura Version

5.7.1 and 5.6.0

### Operating System

Windows 10

### Printer

Creality Ender 3v2 Klipper

### Reproduction steps

1. import a cube from the calibration parts menu
2. scale the cube to 30mm
3. untick uniform scale
4. set Z height to 0.20mm
5. layer height and initial layer height set as 0.20mm
6. slice file

### Actual results

1. file slices 
2. filament values by weight and length are zero
3. time to print is zero
4. file runs home, warm up bed mesh etc the finishes without printing
5. increase the z height of the model from 0.20mm to 0.201mm
6. it slices with a tome value and a length value of 1 minute and 0.08g 

### Expected results

a single layer cube of 30x30mm and 0.20mm heigh

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]

GregValiant, 2024-05-28:
Thanks for the report.
This might have something to do with the plugin.  When importing a regular STL file and setting the model to 0.20 thick, I don't have a problem with it.

The author of the "Calbration Shapes" plugin has removed the plugin from the MarketPlace because of incompatibilities with Cura 5.x versions.  This might be another symptom of that.
Try a regular model (not one of the calibration shapes) and see if this is repeatable on your system.

This cube is 25 x 25 x 0.20 and both the preview and the gcode are correct.  The reason that the print time is 0 minutes is because the print time (as shown in the gcode) is only 46 seconds so it's rounding down to the closest minute which in this case is zero.
;FLAVOR:Marlin
**;TIME:46**
**;Filament used: 0.0592386m**
[image]

MikeOxawopper, 2024-06-03:
Hi,

I did try creating the same print via fusion, I have not tried to print it as yet because the printer is busy.
As for calibration shapes, 1st off I didn’t install it when I did the fresh install of 5.7.1, it pulls it in from somewhere else? 
I had a lot of issues with 5.7.1 having operand errors whenever a file was sliced.

 

I found the only way to fix that was to cut and move all the roaming data files for any other version into a back up folder the install 6.7.1 fresh, then copy the filed for my saved printer profiles and materials into 5.7.1 roaming folder.

Then put all the 5.3.1 roaming data back for that version.

Both versions seem to have imported Calibration shapes from somewhere, it works as expected in 5.3.1 but nothing after.
I’d rather not do a full clean install every time there is an update, exporting and importing profiles every time.

Thanks

 

Sean

 

image.png (view on web) <https://github.com/Ultimaker/Cura/assets/64202104/04d38299-3c89-47b4-9f73-299e2eedf0c4>