Ultimaker/Cura #19156: Movement to the right and back edge before printing

loewilein, 2024-05-28:
### Cura Version

5.71

### Operating System

windows 11

### Printer

LNL3D D3/ Tenlog D3Pro 

### Reproduction steps

Movement to the right and back edge before printing
happens on all parts, sliced with cura 5.71 
this is the problem: 
;LAYER_COUNT:98
M900 K0.050000 T0 ;added by LinearAdvanceSettingPlugin
;LAYER:0
M107 P1
**G0 X21.875 Y6.145 Z0.3**
;TYPE:SKIRT

The problen occures since i installed Cura5.71. No problems with cura 5.5. I used the same printer settings in 5.71 .

### Actual results

No steps to do, becaus it works on older cura versions 

### Expected results

Cura should not generate tge G0 F4800 X294.575 Y274.575 code 

### Add your .zip and screenshots here ⬇️

[cura.zip](https://github.com/Ultimaker/Cura/files/15468055/cura.zip)
[Screenshots.zip](https://github.com/Ultimaker/Cura/files/15468087/Screenshots.zip)

GregValiant, 2024-05-28:
Thanks for the report.
It appears that even with the Prime Tower disabled that the first move is going to the Prime Tower location.
For a fix right now, enable the Prime Tower and change the Prime Tower Location to X25 Y25.  That will disable the move across the build plate.  You can then disable the Prime Tower.
Here is the errant move from 0,0 to the prime tower.
[image]

@HellAholic here is a project file of this one.
[GV 19156.zip](https://github.com/Ultimaker/Cura/files/15472697/GV.19156.zip)

GregValiant, 2024-05-29:
@HellAholic funny you should mention that move.  I scribbled up a post processor for myself that creates a toolpath for the nozzle to move around the periphery of the bed so it gets to the layer start without crossing the print area.
It ended up at 60 lines because of rectangular beds, and rectangular beds with the origin at center, and the G2 G3 moves for elliptical beds, .  It wasn't as simple as I thought it would be.

I have to admit that It never really occurred to me to have the path cross the print area twice.

HellAholic, 2024-05-29:
Added a ticket for investigation, internal reference: CURA-11937

Side notes for Greg :P 
We also have a ticket for the combing in initial move, but as you figured it is very tricky with different printers and build plate shapes and prime positions, initial printing position. the heat up nozzle position, etc. The initial move (simplified version) was aimed to address the issue in the pictures below, for materials such as TPU which are very oozy :) By adding an extra travel move, it keeps out the extra material outside the print area. 

Issue:
[image]
Minor fix by traveling to top right/switch position/prime tower position (in "most" cases)
[image]

loewilein, 2024-05-29:
Thansk for all your help, it works fine now

MariMakes, 2024-07-16:
👋 Quick update on our side.
This is resolved in the 5.8 release 🎉 , you can download the Beta version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.8.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.8.0-beta.1)

I'll close this issue since it's resolved.
Thanks again, and please let us know if you run into any other issues 💪

closed 2024-07-16 by MariMakes