Ultimaker/Cura #19157: 2mm support towers with tree support and small angles

jonyo, 2024-05-28:
### Cura Version

5.7.1

### Operating System

MacOS Sonoma 14.5

### Printer

Creality CR-6 SE

### Reproduction steps

1. Enable tree support
2. Slice it on a model with V and "window" overhangs close to the angle where there isn't much room for trees. Also seems to need things to be an an angle slightly, vertically.

### Actual results

Slices with weird [] columns, some are as small as 1 line of extrusion and about 1-2mm long. The columns even go through actual tree supports in places, that is why I think it is a bug. They seem to "drop straight down" vertically from some support in the print.

I'm not sure if any printer would be able to print a small single line column that is 2mm long and 1000mm in the air. Mine sure can't, sometimes it gets maybe a CM or so before it messes up then has random globs you have to sand off after that.

You can see the vertical "lines" in the screenshot:

[image]

### Expected results

Should slice with no weird 1 line column things.

Note: I've found that if I add a support block in the path of a line column thing, it corrects for it and uses a tree branch instead. On some models there are a ton of lines, so hope this can be fixed.

### Add your .zip and screenshots here ⬇️

[image]

Note: the "actual" ones I'm seeing this on are with STL that are licensed. I was able to reproduce it with a simple shape, but I did have to tilt it in several directions slightly on the build plate to reproduce the lines, so I think it is related to having vertical walls that have a slight angle to them.

[test-support.3mf.zip](https://github.com/Ultimaker/Cura/files/15472768/test-support.3mf.zip)

GregValiant, 2024-05-28:
Thanks for the report.
If you increase the "Maximum Branch Angle" to 70° you should see those additional supports go away.

I'll remove the bug label from this as it is a settings issue.  Let us know if that change works for you.
[image]

jonyo, 2024-05-28:
> Let us know if that change works for you.

Sadly, not for the model I'm actually trying to print. The one uploaded was a "duplicating the bug" effort because the actual STL is part of one of those that you can buy and they say you can't share the STL or anything.

So on the original one, I tried increasing it, even 80 it still had those lines going down:

[image]

 Finally at 85 there were no lines, but at that point it is trying to print almost horizontal branches in some cases, it would need supports for the supports! (which it did not have)

I suspect if you played with the angles and how close to the "v" the "windows" are on the sample part it would get the same issue.

> as it is a settings issue

I'm not following? I totally get, the settings let you do some weird stuff, some things "seem" like a bug at first glance, but really it's just doing what you set in the settings.

I don't see how that is the case here?  It seems like, if you adjust the branch angle such that it cannot "reach" some spot, it should just not try to add a support there.  In fact that is exactly what it does in some cases, branch supports just don't work that well for certain situations. I get that, I had an earlier STL I could not use branch support on, at least not until I modded the STL to add a little horizontal built in connection, then it was able to handle it with a larger area to support.

But why would lowering the branch angle ever result in those weird lines going down?

This feels a little like that old joke: you go to the doctor and tell them when you raise your hand like this, it hurts. The doctor says "then don't raise your hand like that" 😂 

I've played with the settings a fair bit with the actual licensed STL in question.  A few things I've noticed:

- As you pointed out (I did not think to try this one): increasing branch angle makes the number reduce, or go away with high enough angle.  So maybe it happens when it has trouble "reaching" a spot with a branch?  BUT the weird part, sometimes these lines go down from a spot that actually has a branch that reached it (see the last point)
- Adding "support block" anywhere in the little line's path, gets rid of the entire line. I saw one time, the branches re-arranged so that after that a branch successfully reached out to a spot it did not try before. I cannot reproduce it now so there is a chance I'm remembering wrong.
- At one point, I was messing with support % and towers. Maybe this is some bit of "tower" code that is still running even though it is in branch mode?  The reason I'm using tree supports here, is that using normal - it ends up with a bunch of tall towers that my printer has trouble with.  Anyways, so I was messing with settings, and at one point the little lines grew to be larger, and they started resembling those towers it was trying to print in "normal" mode.
- The lines seem to not be needed about 80% of the time... They end up terminating at the top at a spot that looks plenty supported to my eyes.  Maybe it was some artifact of the support it could not reach, just a small little spot the branch could not get to?  To me this smells like an overlap problem in the generation of the supports: Like "part A" says hey this spot needs a support, part B says oh no, that breaks the rules, no support there, but part C didn't get the memo so tries to still fill in support there, and it ends up just dropping a line to the first collision it can find.  This is just a shot in the dark based on observing from the outside, not looking at source, so ¯\_(ツ)_/¯ I can't quite get things to square up. Because it seems like some zig-zag infill that "escaped" maybe due to a bug, BUT when I set infill to 0 it still happens. So just looking from the outside I can't figure it out. Other than like I said, it smells like some kind of overlap / rounding type of issue where different parts of the slicer get it slightly wrong. Anyways hopefully that makes some kind of sense...

jonyo, 2024-05-28:
Also: If you start from "super" quality defaults, turn on support, set support type to branch, and those are only 2 changes from defaults, it still ends up with the lines on this STL. Hmm, in fact it looks like more lines than with the tuning changes I've made over time. Dunno if that helps or not, just thought I'd mention in case it helps.

GregValiant, 2024-05-28:
I can understand not passing out the model.  Put together a project file and keep it off to the side.  Someone form the Cura team may ask for it but you can do that offline so the model doesn't become public.

I'll stick the bug label back on this and the Cura team will take a look.

jonyo, 2024-05-29:
Thanks @GregValiant hopefully the example one is enough so they can maybe see what is causing it, even though it does go away if you raise the branch angle. 

I don't know if this will help diagnose it or not but I noticed something else: as I was trying to make this printable as a work-around... Adding "support erasers" in the path of the weird support lines, when I add an eraser at the very base. Lower "windows" that did not have any visible support lines to them, but did have "in window" supports, the supports went away even though the eraser was at the base, not near the window itself.

The way it behaves, I wonder if it is going through a "first pass" adding support, then a second pass looks for places that violate the setting rules (like not being too close to a wall, etc).  Since the first pass is blocked from trying to add a support to begin with, those windows don't get anything. Normally maybe a first pass would have made a start of a branch but maybe the rules would have taken the branch out but left the support interface?

Also I noticed on the places that have support interface in the window but not even a line going to it, if I zoom in, it looks like the "tip" of a support branch almost:

[image]

GregValiant, 2024-05-29:
When I first say this report I was thinking that it would be "Use Towers" or maybe "Support Horizontal Expansion" that was causing it.  That doesn't appear to be true though.
The team will take a look.
All those little extra supports would fail.  The print would be OK but there might be a lot of "Klingons" as those tiny extrusions didn't stay where they were put and glued themselves to the side of the print when the nozzle moved back to the main print.