Ultimaker/Cura #19159: Support Overhang Slicing Error

MagnumRA07, 2024-05-29:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

Ender 5 Plus

### Name abnormal settings

Support Overhang Angle

### Describe model location

No

### Describe your model

Peg shape for yarn spinner that has flat overhang

### Add your .zip here ⬇️

Tried setting Support Overhang Angle to 90 to get flat overhang and it is not generating supports and is failing to slice.

[image]
[image]

GregValiant, 2024-05-29:
Thanks for the report.
You went the wrong way.  90° is "don't support anything" and 0° is "support everything".  Setting it to 45° should work.
Cura colors the "area needing support" red, and with the setting at 90° you don't have any red in that image.  That's why support isn't being generated.  Just for a test set the Support Overhang Angle to 0° and look at the preview.
The max angle is dependent on the line width and layer height but as a ballpark number the max angle you can print without support is ~63° (measured from vertical).  Once you get over that the individual extrusions are too far apart to bond to each other.

This looks like a setting issue so I'll remove the bug label.  I'll leave this open for a while so you can respond.

MagnumRA07, 2024-05-29:
Thanks for the clarification Greg. I'll need to play with the settings a bit more as I don't particularly like the way the supports are generating, but we can close this one.

GregValiant, 2024-05-29:
That's an excellent model for normal supports with a grid Support Interface at 50% density and a "1 layer height" top distance.
[image]

There are over 70 support settings but don't let them daunt you.  Normal supports will work for a lot of things and are especially effective when you don't need the "reach" of tree support.