Ultimaker/Cura #19161: Bridging setttings over support

ansonl, 2024-05-30:
### Is your feature request related to a problem?

When printing over a support/support interface of another material that does not stick as well, it is useful to print much slower with similar settings used for normal bridging. 

Cura currently uses the normal wall and skin speed for the first layer that prints on top of support. 

Speed preview showing faster normal wall speed over support
[image]

### Describe the solution you'd like

It would be useful to add checkbox to enable the current experimental bridging settings over support. Or have a separate bridging over support setting.

### Describe alternatives you've considered

The bridging setting can be tricked to be on for skin over support only by setting the Bridge Skin Support Threshold to 99%. However this only affects skin and not the walls as seen below

Speed preview
[image]

### Affected users and/or printers

All users who print with supports, especially support interfaces with different materials.
There have been a lot of threads in previous years asking for this feature but I did not find a Issue made for it
https://community.ultimaker.com/topic/33340-layer-speed-above-support/
https://www.reddit.com/r/Cura/comments/i4gqw3/how_to_slow_the_first_layer_above_support/

### Additional information & file uploads

[Support_test.zip](https://github.com/Ultimaker/Cura/files/15493204/Support_test.zip)

GregValiant, 2024-05-30:
This same suggestion was made in 2017 in [Cura Engine #463](https://github.com/Ultimaker/CuraEngine/issues/463) and it was decided not to implement it.  That was before the "Bridging" settings were introduced though.

These are just some thoughts...
The current Bridge settings seem to make it semi-possible to convince Cura to print the skins over support as a bridge feature and use the Bridge Skin speeds.

Here I have the Bridge Skin and Wall speeds both set to 15mm/sec.  The Bridge Skin speed was enforced but not the Wall speed because the walls are not being defined as bridges with my settings.  The Support-Interface Density is 95% and the "Bridge Skin Support Threshold is 96%.
Notice that the right end of the bridged area is supported by the model.  This actually looks pretty good.
[image]

This is another skin on the same model and the same slice.  In this area the "first skin over support" is cantilevered and the Bridge settings were not enforced for this feature.  This is what you normally get anyway.
[image]

I don't know the Cura Engine, but this doesn't look easy to do.  Things like "if the first layer over support is an air gap" will come into play and the exact model geometry above the support seems to come into play.

ansonl, 2024-05-30:
Is there a way to recognize the walls over support interface as bridges in settings? I had the same issue with my workaround and right now I just set the normal outer wall speed lower and hope the border sticks well enough and just live with the much slower overall print time. 

I am printing PETG with a PLA support interface with an interface Z distance of 0. When printing slower over the 2 layer high interface, it results in a smooth breakaway support that has a perfect bottom surface. 

This has good results in PLA printed on a PETG interface and PLA print and stick even better than PETG for the first layer over the support interface.   

[image]

[image]

[image]

smartavionics, 2024-05-30:
Hi @ansonl , my Cura variant which is based on Cura 4, provides a "bridge over support" setting. If you want to try it out, my releases can be found at https://github.com/smartavionics/Cura/releases.

Note to Cura devs, it's trivial to add this setting. This is what I did, back in 2019...
```
diff --git a/src/FffGcodeWriter.cpp b/src/FffGcodeWriter.cpp
index 731ff2583..928f51692 100644
--- a/src/FffGcodeWriter.cpp
+++ b/src/FffGcodeWriter.cpp
@@ -1845,11 +1845,10 @@ void FffGcodeWriter::getBridgeAndOverhangRegions(const SliceDataStorage& storage
 
         const coord_t layer_height = mesh_config.inset0_config.getLayerThickness();
 
-        // if support is enabled, add the support outlines also so we don't generate bridges over support
-
         const Settings& mesh_group_settings = Application::getInstance().current_slice->scene.current_mesh_group->settings;
-        if (mesh_group_settings.get<bool>("support_enable") || mesh_group_settings.get<bool>("support_tree_enable"))
+        if (!mesh.settings.get<bool>("bridge_over_support") && (mesh_group_settings.get<bool>("support_enable") || mesh_group_settings.get<bool>("support_tree_enable")))
         {
+            // add the support outlines so we don't generate bridges over support
             const coord_t z_distance_top = mesh.settings.get<coord_t>("support_top_distance");
             const size_t z_distance_top_layers = round_up_divide(z_distance_top, layer_height) + 1;
             const int support_layer_nr = layer_nr - z_distance_top_layers;
~
~
```

vjvarada, 2024-06-28:
> Hi @ansonl , my Cura variant which is based on Cura 4, provides a "bridge over support" setting. If you want to try it out, my releases can be found at [...]

This feature is something I triedyears back, and @smartavionics build worked really well, was able to print ABS over BVOH: #https://github.com/Ultimaker/Cura/issues/6656

@smartavionics any way to include this into Cura 5.x?

smartavionics, 2024-06-28:
> @smartavionics any way to include this into Cura 5.x?

As I mentioned above, it's a trivial change that would take the UM devs very little time to implement if they thought it was worth doing.

wwmotorsport, 2024-07-22:
I have the exact same issue with trying to print PETG with a PLA support interface layer. I know that custom print properties for the first layer above supports are possible in other slicers, even as far back as the hay days of Simplify3D. I really like Cura, but having just finished a tool changing printer build specifically to do this multi material support interface, not having this control may be a deal breaker....