Ultimaker/Cura #19162: Heavy performance problem when multiply several object

nachorepara, 2024-05-30:
### Cura Version

5.7.1

### Operating System

Windows 11

### Printer

Hellbot Magna 2 300 Dual

### Reproduction steps

1. i drop 3 STL to combine them, thats good so far
2. i combine the items to create a unified object in the bed, 
3. i group the three object int one, to multiply object all together 
4. when yoy press right click and multiply selected object, you can select cuantity, all good so far
5. when you select a number, even it was 2 or 3 and press OK it takes extremely too long to multiply the object
6. if you select to multiply to 20 or 40 items, it can take more than 20 minutes, it is insane
7. it doesn't tkae so long when you multiply not conbined objects
8. i notice that the app consumes only 1 core of my ryzen 3 3600, at 100%, and nothing of GPU
9. when the eternal process finish and multiply is done, you cannot do anything in the app, whitout a long long wait ahead, like move an objet, rotate camera, etc. All takes several minutes from that moment

### Actual results

1. when the eternal process finish and multiply is done, you cannot do anything in the app, whitout a long long wait ahead, like move an objet, rotate camera, etc. All takes several minutes from that moment

### Expected results

we can not lost control of the app when copy elements on surface

### Add your .zip and screenshots here ⬇️

[TT a Cura.zip](https://github.com/Ultimaker/Cura/files/15493793/TT.a.Cura.zip)

GregValiant, 2024-05-30:
Thank you for the report.
I cannot duplicate this on my Win10 Laptop.  I deleted all but one model, and then multiplied it 99 times and it completed in about 15 seconds.  It did take another 30 seconds before I could manipulate the scene, but that is likely a function of my GPU.
Here is the performance graph and 4 cores seem to be involved.  The spikes in the right 4 charts are when Task Manager started.
[image]

This seems to be a system specific problem.  Do you have the Cura priority set to more than 1 core?

nachorepara, 2024-05-30:
Hi!, thanks for your reply. I double check the priority of my cores, and are all selected.
Maybe y missed one step between step 1 and 2, is like this:
1. i drop the models three in the surface
2. i change one of them to select the extruder number 2
3. then i stack all of them to form one piece of "hello kitty" in three different colors
4. combine all of three
5. then all the rest of steps indicated before

i tried in another pc laptop with intel, and do the same, version of cura 5.4

GregValiant, 2024-05-30:
I get the same numbers again.  15 seconds to multiply and about 30 more to be able to manipulate the scene.
My Dell laptop is a unit I purchased used and it is running Win10 Pro with 16mb of RAM.  It is not any type of high-end system.

There have been some problems with some Windows 11 installations and this may be another one.  The Cura team will take a look.

nachorepara, 2024-11-08:
hi there! any news with this issue? thanks!