Ultimaker/Cura #19169: Duplicated objects are printed differently

T3rm1, 2024-05-31:
### Cura Version

5.7.0

### Operating System

Windows 10

### Printer

Elegoo Neptune 3 Pro

### Reproduction steps

Open project file, slice it and preview layer 6.

### Actual results

Some circles have an inner wall.

### Expected results

All objects of the same geometry should be printed the same.

### Add your .zip and screenshots here ⬇️

[image]
[BugCircles.zip](https://github.com/user-attachments/files/15518393/BugCircles.zip)

GregValiant, 2024-05-31:
Thanks for the report.
The walls of all the models are 1.2mm thick.  Your line width is 0.42.  This looks to be a rounding problem as Cura is right at the edge of making two extrusions fit, but on some layers it decides it needs three narrower extrusions.
If you set the Line Width to 0.40 all the models will be the same.

The Cura team will take a look.

This is with the Line Width at 0.40 and it is consistent until it gets to the top radius on the walls and then it drops back down to two wider extrusions.
[image]

T3rm1, 2024-05-31:
Thanks for taking a look so quickly at it. It didn't cause any issues but I figured I'd report anyway as this is a simple example and might fix other problems along the way.
I never understood the line width setting. Cura says it should be 0.4 (like the nozzle) and lower values could produce better results but the default value is 0.42 🤷 And when sliced lines have different widths anyways.

GregValiant, 2024-05-31:
I usually touch up the line width for prints with narrow features.  A 0.4 nozzle can print down to maybe 0.38 (Cura lists the minimum at 0.34 for a .4 nozzle) or up to 0.60.  Cura has a minimum line width for any "Print Line Width" but there is no maximum line width.
Somewhere in there a number is getting rounded up or down differently when it goes through the code.  I don't know if that is something they can fix or not.  At some point the numbers must be rounded from 8 decimal places to two or three and that is going to cause some variance.  Between .40 and .45 something is going on and it involves the wall being 1.2 wide.