Ultimaker/Cura #19174: Two layer skirts

Gphantom650, 2024-06-01:
### Cura Version

5.7.1

### Operating System

Windows 10 Pro

### Printer

FLSUN QQ-S Pro

### Reproduction steps

Sliced a model.  The printer made a skirt, printed layer 1 (0.2mm) then at layer 2 (0.4mm) it printed another skirt.

### Actual results

Second skirt was printed at 2nd layer

### Expected results

Only 1 skirt should be printed before starting to print the model.

### Add your .zip and screenshots here ⬇️

[WoolTensionor-Body.zip](https://github.com/user-attachments/files/15522787/WoolTensionor-Body.zip)

Gphantom650, 2024-06-01:
Not a major bug.

Asterchades, 2024-06-02:
This is by design, and was documented in the update notes for 5.4.0:

> Introduced Skirt Height setting to make a skirt easier to remove

This defaults to 3 globally but can be changed by the printer definition, and can be reconfigured in the print settings. Setting the **Skirt Height** to 1 under Bed Adhesion will restrain it to a single layer.

closed 2024-06-02 by Asterchades

GregValiant, 2024-06-02:
Thanks for the report.
There is a new setting "Skirt Height" and for some reason the default is 3.  I am not a fan.
This is a single line skirt with an exaggerated skirt height of 20.
[image]

I remember the feature request and the reason stated for wanting this was because the person had trouble getting a fingernail under the skirt to remove it from the bed.  Really!!!
I added an over-ride to my definition file to set it to 1.  You can also set it to 1 in the Cura "Bed Adhesion" settings and then save a custom profile so you will always start with a single.

So this is a feature and not a bug.  I'll go ahead and remove the label and close this.

This is your project file and you have the "Skirt Height" set to 3.
[image]