Ultimaker/Cura #19177: Color scheme expansion in print preview

PGTBoos, 2024-06-02:
### Is your feature request related to a problem?

With some printer settings enabled, like continuous infill, it be nice to be able to see the difference between the previous layer and the current layer

### Describe the solution you'd like

Ideally have the previous layer in the same colors but 50% mixed with black or gray.,
Currently when playing a single layer already all below is gray, but when scrolling layers this effect is gone.
An option to have it there, or with darkened colos (as I suggest here) would be nice.
(I wish I could code it myself but I don't know where it lives.)

### Describe alternatives you've considered

I have no alternative, this is mainly a gui think to see things beter.

### Affected users and/or printers

all

GregValiant, 2024-06-02:
Have you used the lower slider control and moved it up?  It will hide the lower layers in the same fashion as the upper slider hides the upper layers.
This is three layers.
[image]

If you click into the "layer number" box you can use the Up Down keys to step through the layers.

PGTBoos, 2024-06-02:
Yes i have i would like to that in layer to layer as well

Outlook voor Android downloaden<https://aka.ms/ghei36>

image.png (view on web)<https://github.com/Ultimaker/Cura/assets/64202104/29546470-bb7d-4323-a8d0-9eb12be96d01>