Ultimaker/Cura #19180: 5.7.2 can't open stl directly from Thingiverse

vini59fr, 2024-06-03:
### Cura Version

5.7.2

### Operating System

Window 10

### Printer

Sovol 01

### Reproduction steps

On Thingiverse, chose a thing, clic on "Open in Cura"

### Actual results

Ultimaker 5.7.2 don't open, only this popup (no options to choose the good software)
[image]

### Expected results

ultimakre Cura should open directly from thingiverse

### Add your .zip and screenshots here ⬇️

.

GregValiant, 2024-06-06:
Thanks for the report.
I get a popup that says "Cura 5.7.0 required".
If you go to the Cura Marketplace and install the "Thingiverse" plugin you can start in Cura and the models will come in to 5.7.2.

I'll leave the bug label on this but I have a feeling it's a website problem.  The code there may need updating.

vini59fr, 2024-06-06:
thank's for reply
There is no "Thingiverse" plugin :(  I just found Thingibrowser.

Still the same issue

gh13, 2024-06-06:
@vini59fr Thingiverse Open in Cura is working for me in 5.7.2 on Windows 10. (Not gloating)
A couple of things I stumbled upon with previous versions.
If you uninstalled a previous 5.7.x version of Cura after installing an newer version it breaks the Thingiverse links to open the STL in the newer version & a reinstall of the newer version is needed to recreate them.
Maybe unrelated, but I recently had an update to the Windows 3D Builder/Viewer, which reset my STL file association so they opened in those apps instead of Cura when an attempt was mad to open locally stored STL file & In had to change the STL file association back to open in Cura. So it might be worth checking that

GregValiant, 2024-06-06:
The ThingiBrowser is working for me as is the direct download from the website to 5.7.2.

Anything that anyone can add in regards to this behavior is appreciated as I can't duplicate it.  Chasing this one down may be tricky.

vini59fr, 2024-06-07:
Great.
Thank you so much @gh13 This was exactly my issue : install 5.7.2 then uninstall 5.7.0 -> thingivers liks were breaks

I reinstall Cura 5.7.2 (above my first installation) and now it works

GregValiant, 2024-06-07:
I'll leave the bug label on this and leave it open.  Maybe it's something that can be addressed.  The Cura team will take a look.

vini59fr, 2024-08-02:
Hey

Still the same issue with 5.8.0 :  install 5.8.8 then uninstall 5.7.2 -> thingivers liks are breaks
have to install a 2nd time the 5.8.0 and it works

nallath, 2024-08-02:
> Hey
> 
> Still the same issue with 5.8.0 : install 5.8.8 then uninstall 5.7.2 -> thingivers liks are breaks have to install a 2nd time the 5.8.0 and it works

Try again. Should be fixed now. The scraper blocking was being a bit overzealous.