Ultimaker/Cura #19181: Нарезка на слои не выполнена

nafanya2, 2024-06-04:
### Cura Version

5.7.2

### Operating System

windows 10

### Printer

zonestar p802qr2

### Name abnormal settings

любые настройки ведут к поломке

### Describe model location

пробовал любое расположение модели

### Describe your model

не получается нарезать модель

### Add your .zip here ⬇️

ни одна модель не нарезается, будь то простой тестовый куб или лодка, настройки задавал все возможные, всегда падает в ошибку
[ебучая ошибка.zip](https://github.com/user-attachments/files/15552469/default.zip)

GregValiant, 2024-06-04:
Thanks for the report.
Try right-clicking on the desktop icon and select "Run as Administrator" just to see if that will work.  It was a problem with some installations of 5.7.0 and 5.7.1.  Let us know if it works.

nafanya2, 2024-06-05:
Попробовал, к сожалению, это не срабатывает
Установил версию 5.4, все работает прекрасно