Ultimaker/Cura #19183: Ultimaker S5 prime tower bug. Prime tower not layiing full perimeters on bed.

JHawk11, 2024-06-04:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

Just using normal prime tower. 
[image]
[image]

### Actual results

Prime tower does not lay a full solid perimeter on the bed. When the prime tower becomes tall, it will detach from the bed with a small amount of stress. 

### Expected results

This is a screen shot from the prime tower on cura 5.2.1 

[image]

[image]

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2024-06-04:
Thanks for the report.
I can reproduce this in 5.7.1 as well.  It does not appear to be S5 specific.

This is an "Interleaved" prime tower.  I've kicked up the purge amount to make it easier to see.
[image]

The only change here is that this is a "Normal" tower.
[image]

They look the same to me and this carries through all the way up.  It appears that the setting has a bug and it isn't being enforced.  That could be having a tumble-down effect and other things are breaking as a result.

@HellAholic and The Cura team will take a look.

[19183 Bug Report.zip](https://github.com/user-attachments/files/15568223/19183.Bug.Report.zip)

3Duke3, 2024-06-20:
Does anyone at UltiMaker actually check changes like this to see if they actually makes a physical improvement, or is it all theoretical? I noticed this behavior after updating to 5.7.1 - this zig zag prime tower behavior just doesn't seem to work for me.

Fortunately, there is a hidden setting pictured below, and if you adjust it to a small value like 0.4 you can get rid of this zig zag rubbish.

[image]

---

However, I came here as I wanted to verify if we are now just printing the prime tower in both materials on every layer?

In this example we have a 1mm thick support interface midway up. But it seems we are building the prime tower with secondary material on every layer leading up to it, even through its not being used to support the part.

[image]