Ultimaker/Cura #19185: I CANT SLICE BIG THINGS, AND I HAVE A VERY GOOD COMPUTER

lavallenfelipe, 2024-06-04:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

Creality Ender 3 V2 NEO

### Describe model location

no

### Describe your model

yes, when i do it 1:2

### Add your .zip here ⬇️

When I do big size files, in all the things, like size of the model or big of mb. it dont let me slice
[CE3V2NEO_CUERPO GRANDE FELI.zip](https://github.com/user-attachments/files/15570991/CE3V2NEO_CUERPO.GRANDE.FELI.zip)

GregValiant, 2024-06-05:
Thanks for the report.
This might be related to your operating system.  There have been some problems on Win11 systems.
In Cura - use the "Help | Show Configuration Folder" command and it will take you to the folder where "Cura.log" is.  Zip that log file and post it here.

This is Cura 5.7.1 on a cheesy Win10 laptop.  From the log:  "Slicing took 178.5 seconds" so about 3 minutes.
[image]

lavallenfelipe, 2024-06-05:
[cura.log.zip](https://github.com/user-attachments/files/15572437/cura.log.zip)

GregValiant, 2024-06-05:
Thanks.
It looks like Cura Engine is crashing:
"Backend exited abnormally with return code 3221225477"
and
"Backend exited abnormally with return code 3221226356"
are both present in the log file.

Someone from the Cura team will take a look.

lavallenfelipe, 2024-06-05:
Thank you, can u tell me then how to fix it, or what I need to do, thanks

GregValiant, 2024-06-05:
One thing has come up but it doesn't always work.
Right-click on the Cura desktop icon and select "Run as Administrator".  If that works then you can go to the "Advanced Properties" for the icon and set it to always Run as Administrator.

Aether1777, 2024-06-07:
hey if you have a 13th / 14th gen intel processor there are some stability issues rn if you are using a gigabyte motherboard you can download the latest bios and reset bios settings to default thats what worked for me so gl 

btw my specs are 
i9 14900k
4090
64 gig ddr5

lavallenfelipe, 2024-06-07:
yes i have a i9 13900k and a gygabite z790 aourus alite ax, i will check that, thanks

Aether1777, 2024-06-07:
Exact same mb as me lol add me on discord ill help you get the drivers installed xeno9118

lavallenfelipe, 2024-06-07:
you solved my problem, now i can slice normally, thank u so much, I fix it

GregValiant, 2024-06-07:
I'll keep this in mind and go ahead and close it since it's a motherboard / config problem.
The Cura team is aware of this happening.  Maybe there will be something they can do from their side.

closed 2024-06-07 by GregValiant

Aether1777, 2024-06-07:
There's nothing cura can do this only affects 13and 14th gen intel cpus most motherboards have power limits set to 4096w which causes instability the only fix for this is to wait patiently for a bios update (or change the settings yourself) [settings](https://videocardz.com/newz/intel-addresses-instability-issues-with-13th-14th-gen-core-k-series-cpus-in-latest-statement#:~:text=According%20to%20HardwareLuxx%2C%20Intel%20has,settings%20applied%20by%20motherboard%20manufacturers)