Ultimaker/Cura #19186: "STL/OBJ file loading takes too long on Linux" Affected: 5.3.0 to 5.7.2

kafufafu, 2024-06-05:
### Cura Version

5.3.0 to 5.7.2

### Operating System

Ubuntu 24.04 /  Dell 5459 core i5 6200u

### Printer

AnetA8 and old printers 

### Reproduction steps

I have discovered a way to replicate the issue:

1) Run any AppImage (from 5.3.0 to 5.7.2).
2) Set AnetA8 as the default printer.
3) Try to open any file using any possible method.
4) Files take up to 20 minutes to load.
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

_The issue does not occur if the default printer before opening the program is different. (for example, Ultimaker S5)._

_I'm unaware if the issue affects Windows or Mac versions. I don't have any means to verify it. However, it definitely affects the AppImage versions used in Linux._

------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

1) I was able to determine that the issue starts in some of the versions 5.3.0. 
(Curiously, the issue does not affect version 5.3.0-alpha+xmas.)

A) 5.3.0 Beta 1 (undetermined as I couldn't run it)
B) 5.3.0 Beta 2 (the issue appears from this version onwards)

### Actual results

It takes up to 20 minutes to open any type of file.

### Expected results

The expected result would be, for example, what version 5.2.2 does ((an instant opening)).

### Add your .zip and screenshots here ⬇️

"There is no additional information."

GregValiant, 2024-06-05:
Thanks for the report and the extra information.
I tried to duplicate this on my Win10 system and it doesn't happen.

Someone from the Cura team will take a look.

bbczeuz, 2024-06-07:
Duplicate of #19104; issue is still open

sausix, 2024-06-08:
Sounds like the USB printing issue. Didn't it make into the issues here yet?

Someone found out the USB printing plugin blocks the loading process. You do not need that plugin if you print from a USB drive, Octo or similar anyway.

gtollini, 2024-06-11:
This was linked on another issue. Here's the [solution I had found](https://github.com/Ultimaker/Cura/issues/16249#issuecomment-2094405826):

> I fixed this by disabling USB printing altogether (I've never had an USB printer to begin with). To do that:
> 
>     * Go to the “Installed” tab of the Marketplace.
> 
>     * Scroll down to USB Printing and click "disable".
> 
>     * Restart Cura.

gtollini, 2024-06-11:
> Thanks for the report and the extra information. I tried to duplicate this on my Win10 system and it doesn't happen.
> 
> Someone from the Cura team will take a look.

@GregValiant it isn't duplicable on Windows because the problematic code only runs if `Platform.isWindows()` returns `False`. See [my fix for this problem](https://github.com/Ultimaker/Cura/pull/19025/commits/13eb9de5916e47b6f36057840eacf72037830673), which just skips the problematic code altogether by changing the `Platform.isWindows()` check to a hard-coded `True`.

kafufafu, 2024-06-11:
> This was linked on another issue. Here's the [workaround I had found](https://github.com/Ultimaker/Cura/issues/16249#issuecomment-2094405826):
> 
> > I fixed this by disabling USB printing altogether (I've never had an USB printer to begin with). To do that:
> > ```
> > * Go to the “Installed” tab of the Marketplace.
> > 
> > * Scroll down to USB Printing and click "disable".
> > 
> > * Restart Cura.
> > ```
@gtollini 
Indeed, that worked (deactivating the USB printing plugin). Thank you for the information; I am truly grateful for the simple solution to this annoying problem ;)

closed 2024-06-11 by kafufafu

PeterM7, 2024-09-26:
Tried your solution on Kubuntu 24.04 running Cura 5.8.1. Cura seemed to load faster but my printing obj. still hasn't loaded.