Ultimaker/Cura #19191: Cura not slicing

Aether1777, 2024-06-06:
### Cura Version

5.7.2

### Operating System

Windows 11, 4090, i9 14900k, 64 gig of ram

### Printer

ender3 v2

### Name abnormal settings

default profile

### Describe model location

no changes to location

### Describe your model

i sliced it on a older laptop using same settings and version

### Add your .zip here ⬇️

[CE3E3V2_FM7UD6WIWMQUV05.zip](https://github.com/user-attachments/files/15604566/CE3E3V2_FM7UD6WIWMQUV05.zip)
I recently had to reset my pc but before that when slicing it would cause a BSOD

GregValiant, 2024-06-06:
Thanks for the report.
The model doesn't appear to have any errors and it sliced fine on my Win10 system.
The Cura team will take a look.
When you get a minute, recreate the problem in Cura and then use the "Help | Show Configuration Folder" command.  In that config folder will be "Cura.log".  Zip that log file and post the zip folder here.  Maybe it will have a clue.
In 4 years of doing triage here, this is the first bug report I've seen where a Cura crash also crashed Windows to a blue screen.

Your current support settings are using a lot of material and taking a long time.  When you get it to slice you can knock a couple of hours off the print time by considering some changes.
Support Overhang Angle - 60°
Support Density - 10%
Support Infill Layer Thickness - 2X Layer Height (0.40)
Support Distance Priority - Z overrides XY
Minimum Support Interface Area - 1.0

[image]

Aether1777, 2024-06-07:
[logs.zip](https://github.com/user-attachments/files/15696225/logs.zip)
here are the logs

Aether1777, 2024-06-07:
Ok i figured it out there was a issue with intel 13th and 14th gen cpus that caused instability and crashes a bios update by gigabyte that literally just released fixed that problem i installed and revert to defaults and everything works now thanks!

GregValiant, 2024-06-07:
Thanks for that.
I'll go ahead and close this.

@HellAholic this is something I hadn't heard.

closed 2024-06-07 by GregValiant

HellAholic, 2024-06-07:
@GregValiant technically you can get any program to cause a BSOD by overclocking the system to an unstable state, but it's then  the system instability that is causing the issue rather than the program. It is a fun side of the overclocking hobby - if you will - that comes with building computers. It's entirely out of the context of 3D printing but using a benchmark tool for CUP & GPU after doing bios updates/changes or overclocking is usually best practice.
But thanks for the poke, I'll keep this in mind as well :)

GregValiant, 2024-06-07:
This is similar to #19185 and the same fix (update to bios) worked to eliminate the problem.

Aether1777, 2024-06-07:
Yep ik (im kinda the person that figured it out)