Ultimaker/Cura #19192: Building Cura from source in Windows running fails: DLL load failed while importing pyArcus

ggtechd, 2024-06-06:
### Cura Version

5.7.0/5.8.0-alpha.0  

### Operating System

Windows 11

### Printer

klipper supported

### Reproduction steps

pip install conan==1.60.2
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force

cd D:\Cura
git clone https://github.com/Ultimaker/Cura.git 
cd D:\Cura\Cura
conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
.\venv\Scripts\activate.ps1
python cura_app.py

### Actual results

 PS D:\Cura> python cura_app.py
Error in sys.excepthook:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "D:\Cura\cura\CuraApplication.py", line 41, in <module>
    from UM.Qt.QtApplication import QtApplication  # The class we're inheriting from.
  File "C:\.conan\e63eb3b\1\site-packages\UM\Qt\QtApplication.py", line 21, in <module>
    from UM.Backend.Backend import Backend #For typing.
  File "C:\.conan\e63eb3b\1\site-packages\UM\Backend\Backend.py", line 11, in <module>
    from UM.Backend.SignalSocket import SignalSocket
  File "C:\.conan\e63eb3b\1\site-packages\UM\Backend\SignalSocket.py", line 4, in <module>
    import pyArcus as Arcus
ImportError: DLL load failed while importing pyArcus: The specified module could not be found.

Original exception was:
Traceback (most recent call last):
  File "D:\Cura\cura_app.py", line 197, in <module>
    from cura.CuraApplication import CuraApplication
  File "D:\Cura\cura\CuraApplication.py", line 41, in <module>
    from UM.Qt.QtApplication import QtApplication  # The class we're inheriting from.
  File "C:\.conan\e63eb3b\1\site-packages\UM\Qt\QtApplication.py", line 21, in <module>
    from UM.Backend.Backend import Backend #For typing.
  File "C:\.conan\e63eb3b\1\site-packages\UM\Backend\Backend.py", line 11, in <module>
    from UM.Backend.SignalSocket import SignalSocket
  File "C:\.conan\e63eb3b\1\site-packages\UM\Backend\SignalSocket.py", line 4, in <module>
    import pyArcus as Arcus
ImportError: DLL load failed while importing pyArcus: The specified module could not be found.

PS D:\Cura> echo $env:PYTHONPATH
D:\Cura\venv\Lib\site-packages;;C:\.conan\b4d6e58f6d8\1\lib;C:\.conan\088d6872bc8\1\lib;C:\.conan\9dad6953524\1\lib;C:\.conan\55c4e9a6c\1\lib\pyDulcificum;C:\.conan\e63eb3b\1\site-packages;C:\.conan\e63eb3b\1\plugins
PS D:\Cura>

### Expected results

Runned cura application

### Add your .zip and screenshots here ⬇️

[image]
[image]
[cura_coding.txt](https://github.com/user-attachments/files/15605552/cura_coding.txt)

closed 2024-06-08 by ggtechd

zhou123aw, 2024-06-13:
Have you solved this problem? I am also in the same situation

ManuRamesha, 2024-06-22:
Has this issue been resolved? The same thing is true for me as well.

srlihg66, 2024-06-23:
Have you solved this problem? I am also in the same situation too

vjvarada, 2024-06-27:
Same problem here, hopefully, there is a solution for this.

srlihg66, 2024-06-28:
这个问题我已经解决 找到文件路径 pyArcus.pyd Arcus.dll  去安装个.exe 去路径里面把这两个文件 放到pyArcus.pyd  路径下 替换原文件  就可以工作了

vjvarada, 2024-06-28:
Thanks for your help, Replacing those files worked.  I wonder if there is a way to fix the build process so we don't have to do this with every new build.

vjvarada, 2024-06-28:
Even though Cura stars up, It crashes when loading more than one file.  I assume Next2d uses arcus so it throws this error.
```
Thread 0x0000863c (most recent call first):
  File "C:\.conan\6d1c0b\1\bin\lib\threading.py", line 320 in wait
  File "C:\.conan\6d1c0b\1\bin\lib\threading.py", line 460 in acquire
  File "C:\.conan\de57bd\1\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\.conan\de57bd\1\site-packages\UM\JobQueue.py", line 124 in run
  File "C:\.conan\6d1c0b\1\bin\lib\threading.py", line 1009 in _bootstrap_inner
  File "C:\.conan\6d1c0b\1\bin\lib\threading.py", line 966 in _bootstrap

Current thread 0x00004d2c (most recent call first):
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura\Arranging\Nest2DArrange.py", line 123 in findNodePlacement
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura\Arranging\Nest2DArrange.py", line 134 in createGroupOperationForArrange
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura\Arranging\Arranger.py", line 25 in arrange
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura\CuraApplication.py", line 2188 in _readMeshFinished
  File "C:\.conan\de57bd\1\site-packages\UM\Signal.py", line 357 in __performEmit
  File "C:\.conan\de57bd\1\site-packages\UM\Signal.py", line 328 in __performEmitIndirect
  File "C:\.conan\de57bd\1\site-packages\UM\Event.py", line 218 in call
  File "C:\.conan\de57bd\1\site-packages\UM\Qt\QtApplication.py", line 506 in event
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura\CuraApplication.py", line 1223 in event
  File "C:\.conan\de57bd\1\site-packages\UM\Qt\QtApplication.py", line 420 in exec
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura\CuraApplication.py", line 953 in run
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura_app.py", line 243 in <module>
```

vjvarada, 2024-06-30:
found the fix, mentioned here: https://github.com/Ultimaker/Cura/issues/19299

runruiliu, 2024-08-15:
> 这个问题我已经解决了 找到文件路径 pyArcus.pyd Arcus.dll 去安装个.exe 去路径里面把这两个文件放到 pyArcus.pyd 路径下 替换原文件就可以工作了

您好，我也遇到了相同的问题，我在Cura\venv\Lib\site-packages中找到了pyArcus.pyd，Arcus.dll两个文件，接下来我应该怎样安装.exe文件，安装什么.exe文件呢？期待您的回复，不胜感激