Ultimaker/Cura #19198: Cura Doesn't Working

burakturnaoglu, 2024-06-06:
### Cura Version

5.7.2 - 5.6.0

### Operating System

Windows 10 Pro 22H2

### Printer

Ender 3 S1 Pro

### Reproduction steps

I installed Cura version 5.7.2, but after the application loads, it only shows the splash screen. Whether I choose "Log in" or "Skip," the application closes itself. I tried deleting the files located in the %user% part of the program and reinstalling it. I also tried installing it to a different location and on a different drive, but the result didn't change. I set the graphics card option to NVIDIA to run the program, but it still didn't change the outcome. I checked that my graphics card drivers are up-to-date. I need your help.

### Actual results

It's not working

### Expected results

Maybe it can work

### Add your .zip and screenshots here ⬇️

I am sharing a screen recording of the issue I am experiencing with a zip file. I need your help.

[Cura.zip](https://github.com/user-attachments/files/15685460/Cura.zip)
[Cura.zip](https://github.com/user-attachments/files/15685460/Cura.zip)

HellAholic, 2024-06-12:
Hey @burakturnaoglu , did you  try removing the files in both local and roaming?
so %user%/appdata/roaming/cura/... and %user%/appdata/local/cura/...
If that doesn't work, go to the %user%/appdata/roaming/cura/5.7/... and copy the cura.log file, zip that one and add it to the ticket.

burakturnaoglu, 2024-06-12:
[cura log.zip](https://github.com/user-attachments/files/15803812/cura.log.zip)
Here is the log file

burakturnaoglu, 2024-06-29:
@HellAholic is there any opinion?

vo-omega-001, 2024-07-26:
Hi,
I reproduice the same issue in docker container based on mini debian 12,
from Cura github sources: SHA-1: https://github.com/Ultimaker/Cura/commit/935109680357710057f8bc1b6cf2dec3e9d317da

I commented the screens of all_pages_list list in in Cura/cura/UI/WelcomePagesModel.py, to bypass the crash on Skip action.
However I reproduce a close issue when I try to add a printer from Cura GUI.

NB: In welcome page list, the next page after SignIn ( named  "cloud") is "add_network_or_local_printer".