Ultimaker/Cura #19200: Wrong processing of printer start g-code

lucwuyts, 2024-06-07:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

Sovol SV04 with Klipper

### Reproduction steps

Using Cura 5.4.0, this printer start-code 
`M190 S0
M109 S0
START_PRINT EXTRUDER=0 EXTRUDERS={extruders_enabled_count} EXTRUDER_TEMP={material_print_temperature, 0} EXTRUDER1_TEMP={material_print_temperature, 1} BED_TEMP={material_bed_temperature} AREA_START=%MINX%,%MINY% AREA_END=%MAXX%,%MAXY%`

translates to:
`;FLAVOR:Marlin
;TIME:1838
;Filament used: 1.55973m
;Layer height: 0.3
;MINX:137.1
;MINY:137.1
;MINZ:0.2
;MAXX:164.9
;MAXY:164.9
;MAXZ:20
;POSTPROCESSED
;Generated with Cura_SteamEngine 5.4.0
M82 ;absolute extrusion mode
M190 S0
M109 S0
START_PRINT EXTRUDER=0 EXTRUDERS=1 EXTRUDER_TEMP=235 EXTRUDER1_TEMP=210 BED_TEMP=70.0 AREA_START=137.1,137.1 AREA_END=164.9,164.9`

Using Cura 5.7.2 this is not the case anymore and translates to:
`;FLAVOR:Marlin
;TIME:1539
;Filament used: 1.0956m
;Layer height: 0.2
;MINX:136.5
;MINY:136.5
;MINZ:0.3
;MAXX:165.5
;MAXY:165.5
;MAXZ:19.9
;TARGET_MACHINE.NAME:SV04 Single Mode 01
;Generated with Cura_SteamEngine 5.7.2
M104 S200
M105
M109 S200
START_PRINT EXTRUDER=0 EXTRUDERS={extruders_enabled_count} EXTRUDER_TEMP={material_print_temperature, 0} EXTRUDER1_TEMP={material_print_temperature, 1} BED_TEMP={material_bed_temperature} AREA_START=%MINX%,%MINY% AREA_END=%MAXX%,%MAXY%
`

I think this was also the case sinds Cura 5.5.x

### Actual results

`;FLAVOR:Marlin
;TIME:1539
;Filament used: 1.0956m
;Layer height: 0.2
;MINX:136.5
;MINY:136.5
;MINZ:0.3
;MAXX:165.5
;MAXY:165.5
;MAXZ:19.9
;TARGET_MACHINE.NAME:SV04 Single Mode 01
;Generated with Cura_SteamEngine 5.7.2
M104 S200
M105
M109 S200
START_PRINT EXTRUDER=0 EXTRUDERS={extruders_enabled_count} EXTRUDER_TEMP={material_print_temperature, 0} EXTRUDER1_TEMP={material_print_temperature, 1} BED_TEMP={material_bed_temperature} AREA_START=%MINX%,%MINY% AREA_END=%MAXX%,%MAXY%
`

### Expected results

`;FLAVOR:Marlin
;TIME:1539
;Filament used: 1.0956m
;Layer height: 0.2
;MINX:136.5
;MINY:136.5
;MINZ:0.3
;MAXX:165.5
;MAXY:165.5
;MAXZ:19.9
;TARGET_MACHINE.NAME:SV04 Single Mode 01
;Generated with Cura_SteamEngine 5.7.2
M104 S200
M105
M109 S200
START_PRINT EXTRUDER=0 EXTRUDERS=1 EXTRUDER_TEMP=235 EXTRUDER1_TEMP=210 BED_TEMP=70.0 AREA_START=137.1,137.1 AREA_END=164.9,164.9`

### Add your .zip and screenshots here ⬇️

no zip, no screenshots provided

GregValiant, 2024-06-07:
Thanks for the report.
To my knowledge there have never been replacement patterns "MAXX", "MAXY" (or "MAXZ") in Cura.
In addition:  "%" signs are not valid enclosures in Cura.  Cura requires "curly brackets" to enclose "Replacement Patterns".
It is likely that you had a post-processor or the Klipper plugin running that read the previous "MAX" lines and made the insertions.

I'm going to remove the bug label from this and close it as it is not a Cura problem.  Let me know if you feel that is premature.

closed 2024-06-07 by GregValiant

lucwuyts, 2024-06-08:
Thank you.
You are right about the MAX patterns, but it did not insert the correct temperatures for some reason.
I reinstalled Cura, and for the moment it is working correctly.   
(I did not install my extra plugin yet for the replacement of the MAX values, but the temperatures are correct now.