Ultimaker/Cura #19203: [5.7.2] Method X generic materials not available with LABS Extruder

cdsomerville, 2024-06-08:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

Method X

### Reproduction steps

1. Specified LABS extruder for extruder 1
2. Clicked "Enabled"
3. Generic Materials greyed out
4. Can only select between Ultimaker materials
5. Downloaded numerous material profiles from Marketplace
6. Restarted Cura
7. Still unable to access any generic profile

### Actual results

You are unable to specify generic materials for use on the LABS extruder. Only Ultimaker materials can be selected. I have not tried creating new profiles tuned for alternate materials yet but I expect the same result. 

### Expected results

One should be able to access all downloaded 3rd party filament profiles or access generic printing profiles. 

### Add your .zip and screenshots here ⬇️

NA

GregValiant, 2024-06-08:
Thanks for the report.

The Makerbot merger is a work in progress.  Ultimaker does a lot of material testing and it's possible that those other materials did not get tested prior to adding the MethodX to the supported printer list.

Maybe there is something you can do now.
Go to the Cura installation folder and the definitions sub-folder.  It should be something like:
_"C:\Program Files\UltiMaker Cura 5.7.2\share\cura\resources\definitions"_
Open the "ultimaker_methodx.def.json" file in a text editor.  You will see a lot of materials in the "exclude_materials" section.
You can remove the ones you don't want left out.  I would assume that would be all the generic types.
Use do care.  All the lines in the "exclude_materials" section must end with a comma except for the last line.

**This...**
"fdplast_",
"filo3d_",
"generic_asa_175",
"generic_abs_175",
"generic_bvoh_175",
"generic_petg_175",
"generic_pla_175",
"generic_tough_pla_175",
"generic_pva_175",
"generic_cffpa_175",
"generic_cpe_175",
"generic_nylon_175",
"generic_hips_175",
"generic_pc_175",
"generic_tpu_175",
"goofoo_",
"ideagen3D_",
**would become this...**
"fdplast_",
"filo3d_",
"goofoo_",
"ideagen3D_",
And the generic materials should show up when you restart Cura.
The disclaimer is that I'm not an UltiMaker guy so your switch to the LABS extruder may well have side-effects that are beyond my limited knowledge.  If you are using the base MethodX definition file, the above may well work.

This is the second report of problems with the MethodX defintion.  I'll mark this as a duplicate and the Cura team will take a look.

cdsomerville, 2024-06-12:
Thank you, this did address the issue. I now have generic profiles available to me in the Method X workspace. I'm sure all this will improve with time.

7enso, 2024-07-03:
Thanks @GregValiant  - That hack worked for me as well for generic materials.

Is it also possible to apply this to specific materials somehow? For example

https://marketplace.ultimaker.com/app/cura/materials/rlsijlbing/BASF_Ultrafuse_PET

is not marked as compatible with Method X and therefore cannot be selected with a Method X printer even tho I deleted all entrys in the exclude_materials section.

Can I somehow make this work?

The BASF Ultrafuse PET is installed but it's not in "\share\cura\resources\materials".

I might also write to BASF Forward AM but for them to add the Method X as a compatible printer might be a bigger step ...

PET might be easy to use from generic profiles but there are others that we would also like to use in Cura.

GregValiant, 2024-07-03:
That's probably part of the "Work in Progress" thing.  I've see a couple of pull requests submitted by UltiMaker employees in regards to different materials.  UM does a lot of work coming up with profiles that work with their products.

If you load the "Material Settings" plugin from the MarketPlace you might be able to change the settings of a Custom material to create what you need.