Ultimaker/Cura #19205: Infill Over 100%

chopperrider, 2024-06-09:
### Cura Version

5.1.1

### Operating System

Window 10

### Printer

Anycubic Kobra 2

### Reproduction steps

The infill was set to 400% by mistake

### Actual results

My 3D printer did not limit the infill to 100% resulted in overflow of the filament and busted the hot end.

### Expected results

The slicer should limit the infill to 100%

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-06-09:
Thanks for the report.
Not all infill patterns fill a model at 100% density.  For example "Lightning" infill can be printed to 240% before the line width gets to 0.40mm, and Grid infill can be printed to 150% density before the line distance gets to 0.40.
There is a warning (the setting box turns yellow) when the setting is > 100% but yes, it will slice.  Depending on the Infill Pattern (and the Line Width), it might not over-extrude.

I'll leave the bug label o this and the Cura team will take a look.  I see you were using 5.1.1 but in 5.7.1 (much better BTW) the situation is unchanged.  There is a warning for "infill_sparse_density" > 100% but there is no maximum value that would cause the setting box to turn red and create a "Don't Slice" situation.

The mess on your hot end should not have been caused by an over-extrusion.  The pressure arm of the extruder is designed to be adjustable and when properly adjusted it will allow the extruder to "skip steps" in over-pressure situations.  That is its function.

Although there can be some leakage around a nozzle that can lead to a large lump forming on the hot end, that depends on how tight the nozzle was, and what condition the threads in the hot end are.  The nozzle is relatively hard brass and the heat block is very soft aluminum.  It's not a good combination for thread longevity.

GregValiant, 2024-06-09:
You can add an override to your printer definition file and it will prevent this from happening again.

The AnyCubic Kobra 2 printer definition file is in the Cura installation folder and the definitions sub-folder.  On my Win10 system it is:
_"C:\Program Files\UltiMaker Cura 5.7.2\share\cura\resources\definitions"_
In Cura - use the _"Help | Show Configuration Folder"_ command in Cura.  In the configuration folder will be another "definitions" folder but it is meant for custom files.  It will be the Save-As folder.

Open "anycubic_kobra2.def.json" in a text editor.
Do a Save-As to the definitions folder within your Configuration folder.

Once you have it saved, scroll down to the end (which is the end of the "overrides" section).  You will see:
[image]

Add a comma to the end of the "wall_thickness" line and then a carriage return and add:

"infill_sparse_density": { "maximum_value": 100}

There is no comma at the end of that line.  Follow the spacing as it is in the other lines.  DO NOT mix tabs and spaces in the indent of the new line.  That would keep the file from loading.

Save the changed file and it will load in preference to the stock file when you start Cura.  You will get this if you make a typo:
[image]

This is something you can do now instead of waiting months for a fix that might or might not get in.