Ultimaker/Cura #19210: Loading models

oinkoink03, 2024-06-10:
### Cura Version

5.7.1

### Operating System

windows 10

### Printer

Ender 3 pro glass bed cr touch mods

### Reproduction steps

1. Opened Cura
2. Load any model

### Actual results

For me it scaled up the model to where it stopped fitting on the print bed. from 1 inch to 25 inches

### Expected results

model should scale up

### Add your .zip and screenshots here ⬇️

If it's settings I have accidentally changed please let me know

oinkoink03, 2024-06-10:
> Thanks. I don't understand.

What part do you not understand?

GregValiant, 2024-06-10:
Post project file as you were asked on the initial page you filled out.  I have no idea how I am to recreate this as I don't have any trouble loading models.  If they are in Imperial units, they get scaled up and I have to re-scale in Cura.  If they are metric they come in as drawn.