Ultimaker/Cura #19212: The temperature cannot exceed 360 degrees. How can we set the temperature more freely

magicmaker3, 2024-06-10:
### Is your feature request related to a problem?

I need to print my own modulated peek, with a maximum temperature of 460 degrees Celsius. However, in the current version, if the temperature exceeds 360 degrees Celsius, it cannot be sliced. I have tried creating a new material, the default temperature is set to 420 degrees Celsius, but it still cannot be sliced at temperatures above 360 degrees Celsius.

### Describe the solution you'd like

Make temperature settings more flexible

### Describe alternatives you've considered

Easy to solve problem, I hope there are not too many restrictions. Thank you for browsing

### Affected users and/or printers

Users with the same needs

### Additional information & file uploads

[image]

GregValiant, 2024-06-10:
This has been discussed previously and likely won't be changed as it is meant as a safety for 99.99% of the printers out there. 

If it is possible to override the High Temperature Limit you may need to have a side conversation with a developer.
Report #12182 may be of use.
A couple of things in regards to that workaround:
"cool_min_temperature" must also be overridden if it is a setting in the Cura version you are using.

Using the workaround described I was able to set the temperatures, and it would slice, but the settings remain "red".  If I altered any temperature setting then slicing would fail.  I could re-open Cura and slicing would be OK until I once again adjusted a temperature.  It appears that something else needs to be changed/overridden.

In 4.13.1 it was all good.  The temperature settings were yellow indicating they were in the warning area.  The temperature settings could be adjusted without issue.

magicmaker3, 2024-06-13:
> With these lines added to the printer definition file "overrides" section you can set the print temperature higher. This will work in versions above Cura 5.0. (I was missing the last two when I tried earlier).
>
> Max.Temp.Override.png (view on web)

thank you

magicmaker3, 2024-06-13:
Confirmed that the issue of children was not taken into consideration,But there is an ancient Chinese saying that goes "give up eating because of choking".  The excellent material PEEK has been completely abandoned by Cura.
Moreover, A typical commercial printer, even at full power, cannot heat up to 360 ℃ or above. A printer that can heat up to such a high temperature is usually not damaged by a temperature increase of 360 ℃.

magicmaker3, 2024-06-13:
Confirmed that the issue of children was not taken into consideration,I know there are other ways to solve it, but I believe that excellent PEEK should not be abandoned.There is an ancient Chinese saying that goes "give up eating because of choking". 
Moreover, A typical commercial printer, even at full power, cannot heat up to 360 ℃ or above. A printer that can heat up to such a high temperature is usually not damaged by a temperature increase of 360 ℃.

magicmaker3, 2024-06-13:
I know there are other ways to solve it, but I believe that excellent PEEK should not be abandoned. There is an ancient Chinese saying that goes "give up eating because of choking".
In addition, general commercial printers cannot heat up to 360 ℃ or above even at full power. Printers that can be heated to such high temperatures are usually not damaged by a temperature increase of 360 ℃.

GregValiant, 2024-06-13:
I don't think it's an "abandoning PEEK" thing but rather that lawyers were involved in the under-writing decision.  At the end of the day, it's a liability issue.

I also scribbled up a post-processor to address this without having to override any Cura settings in the printer definition file.  I felt it necessary to add my own disclaimers.
In Cura you would set any/all temperature settings for PEEK to exactly 1/2 of what you really want and then the script goes through in post-process and for a single extruder printer:
Doubles all the hot end temperatures in the gcode.
Or for a dual-extruder printer:
Doubles the temperatures for either T0, or T1, or Both.

Let me know if you want it.  It isn't as neat and clean as the overrides in the definition file, but it is easier to turn off.

magicmaker3, 2024-07-07:
Thank you for your reply. I understand that you need to consider more. The version 4.4 I am currently using has also solved the problem

GregValiant, 2024-07-07:
Since there are already ways to override the 365° temperature limit, I'll go ahead and close this.  It's a "niche" situation and I don't see the developers making changes to Cura's defaults that would apply to "all" printers when simple overrides in a definition file can do it for a specific printer.
I'll post them again if you want to upgrade to a later version of Cura.  These overrides increase the "Max Bed Temp" to 180 and the "Max Print Temp" to 500.

"material_bed_temperature_layer_0": { "maximum_value": 180, "maximum_value_warning": 130},
"material_bed_temperature": { "maximum_value": 180, "maximum_value_warning": 130},
"material_print_temperature_layer_0": { "maximum_value": 500, "maximum_value_warning": 460},
"material_print_temperature": { "maximum_value": 500, "maximum_value_warning": 460},
"material_initial_print_temperature": { "maximum_value": 500},
"material_final_print_temperature": { "maximum_value": 500},
"material_standby_temperature": { "maximum_value": 500},
"material_break_temperature": { "maximum_value": 500},
"material_break_preparation_temperature": { "maximum_value": 500},
"cool_min_temperature": { "maximum_value": 500},
"print_temp_warn_limit": { "value": 4.60},
"print_temp_anomaly_limit": { "value": 5.00}