Ultimaker/Cura #19217: conan install failed

zhou123aw, 2024-06-11:
### Cura Version

5.4.0

### Operating System

Windows

### Printer

Ender 3 V2 Neo

### Reproduction steps

conan install . --build=missing --update -o cura:devtools=True -g PyCharmRunEnv

### Actual results

ninja: build stopped: subcommand failed.
curaengine/5.8.0-alpha.0+f4ddf7@ultimaker/testing:
curaengine/5.8.0-alpha.0+f4ddf7@ultimaker/testing: ERROR: Package 'aa8f4086cd13978762ab88417d8ed09b4991d9d1' build failed
curaengine/5.8.0-alpha.0+f4ddf7@ultimaker/testing: WARN: Build folder C:\.conan\c4da05\1\build\Release
ERROR: curaengine/5.8.0-alpha.0+f4ddf7@ultimaker/testing: Error in build() method, line 191
        cmake.build()
        ConanException: Error 1 while executing cmake --build "C:\.conan\c4da05\1\build\Release" -- -j20

### Expected results

ok

### Add your .zip and screenshots here ⬇️

[image]

zhou123aw, 2024-06-11:
Cura Version

5.7.2

runruiliu, 2024-08-13:
I also encountered the same problem, may I ask if you have solved this problem now? Could you give me some suggestions on how to solve the problem? Thank you very much